xtools-at/AssistantPi #1: Error when running setup.sh

MarMed, 2017-05-06:
I'm getting this error when running setup.sh

`checking for memmove... yes
checking for memset... yes
checking for mkfifo... yes
checking for pow... yes
checking for select... yes
checking for socket... yes
checking for stpcpy... yes
checking for strcasecmp... yes
checking for strchr... yes
checking for strdup... yes
checking for strerror... yes
checking for strstr... yes
checking for strtol... yes
checking for strtoul... yes
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating man/Makefile
config.status: creating scripts/shairport-sync.service
config.status: creating scripts/shairport-sync
config.status: creating config.h
config.status: executing depfiles commands
make  all-recursive
make[1]: Entering directory '/tmp/shairport-sync'
Making all in man
make[2]: Entering directory '/tmp/shairport-sync/man'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/tmp/shairport-sync/man'
make[2]: Entering directory '/tmp/shairport-sync'
gcc -DHAVE_CONFIG_H -I.  -Wno-multichar -DSYSCONFDIR=\"/usr/local/etc\"  -Wno-multichar -DSYSCONFDIR=\"/usr/local/etc\"  -g -O2 -MT shairport.o -MD -MP -MF .deps/shairport.Tpo -c -o shairport.o shairport.c
mv -f .deps/shairport.Tpo .deps/shairport.Po
gcc -DHAVE_CONFIG_H -I.  -Wno-multichar -DSYSCONFDIR=\"/usr/local/etc\"  -Wno-multichar -DSYSCONFDIR=\"/usr/local/etc\"  -g -O2 -MT rtsp.o -MD -MP -MF .deps/rtsp.Tpo -c -o rtsp.o rtsp.c
mv -f .deps/rtsp.Tpo .deps/rtsp.Po
gcc -DHAVE_CONFIG_H -I.  -Wno-multichar -DSYSCONFDIR=\"/usr/local/etc\"  -Wno-multichar -DSYSCONFDIR=\"/usr/local/etc\"  -g -O2 -MT mdns.o -MD -MP -MF .deps/mdns.Tpo -c -o mdns.o mdns.c
mv -f .deps/mdns.Tpo .deps/mdns.Po
gcc -DHAVE_CONFIG_H -I.  -Wno-multichar -DSYSCONFDIR=\"/usr/local/etc\"  -Wno-multichar -DSYSCONFDIR=\"/usr/local/etc\"  -g -O2 -MT mdns_external.o -MD -MP -MF .deps/mdns_external.Tpo -c -o mdns_external.o mdns_external.c
mv -f .deps/mdns_external.Tpo .deps/mdns_external.Po
gcc -DHAVE_CONFIG_H -I.  -Wno-multichar -DSYSCONFDIR=\"/usr/local/etc\"  -Wno-multichar -DSYSCONFDIR=\"/usr/local/etc\"  -g -O2 -MT common.o -MD -MP -MF .deps/common.Tpo -c -o common.o common.c
mv -f .deps/common.Tpo .deps/common.Po
gcc -DHAVE_CONFIG_H -I.  -Wno-multichar -DSYSCONFDIR=\"/usr/local/etc\"  -Wno-multichar -DSYSCONFDIR=\"/usr/local/etc\"  -g -O2 -MT rtp.o -MD -MP -MF .deps/rtp.Tpo -c -o rtp.o rtp.c
mv -f .deps/rtp.Tpo .deps/rtp.Po
gcc -DHAVE_CONFIG_H -I.  -Wno-multichar -DSYSCONFDIR=\"/usr/local/etc\"  -Wno-multichar -DSYSCONFDIR=\"/usr/local/etc\"  -g -O2 -MT player.o -MD -MP -MF .deps/player.Tpo -c -o player.o player.c
mv -f .deps/player.Tpo .deps/player.Po
gcc -DHAVE_CONFIG_H -I.  -Wno-multichar -DSYSCONFDIR=\"/usr/local/etc\"  -Wno-multichar -DSYSCONFDIR=\"/usr/local/etc\"  -g -O2 -MT alac.o -MD -MP -MF .deps/alac.Tpo -c -o alac.o alac.c
mv -f .deps/alac.Tpo .deps/alac.Po
gcc -DHAVE_CONFIG_H -I.  -Wno-multichar -DSYSCONFDIR=\"/usr/local/etc\"  -Wno-multichar -DSYSCONFDIR=\"/usr/local/etc\"  -g -O2 -MT audio.o -MD -MP -MF .deps/audio.Tpo -c -o audio.o audio.c
mv -f .deps/audio.Tpo .deps/audio.Po
gcc -DHAVE_CONFIG_H -I.  -Wno-multichar -DSYSCONFDIR=\"/usr/local/etc\"  -Wno-multichar -DSYSCONFDIR=\"/usr/local/etc\"  -g -O2 -MT mdns_avahi.o -MD -MP -MF .deps/mdns_avahi.Tpo -c -o mdns_avahi.o mdns_avahi.c
mv -f .deps/mdns_avahi.Tpo .deps/mdns_avahi.Po
gcc -DHAVE_CONFIG_H -I.  -Wno-multichar -DSYSCONFDIR=\"/usr/local/etc\"  -Wno-multichar -DSYSCONFDIR=\"/usr/local/etc\"  -g -O2 -MT audio_alsa.o -MD -MP -MF .deps/audio_alsa.Tpo -c -o audio_alsa.o audio_alsa.c
mv -f .deps/audio_alsa.Tpo .deps/audio_alsa.Po
gcc -DHAVE_CONFIG_H -I.  -Wno-multichar -DSYSCONFDIR=\"/usr/local/etc\"  -Wno-multichar -DSYSCONFDIR=\"/usr/local/etc\"  -g -O2 -MT audio_pipe.o -MD -MP -MF .deps/audio_pipe.Tpo -c -o audio_pipe.o audio_pipe.c
mv -f .deps/audio_pipe.Tpo .deps/audio_pipe.Po
g++  -g -O2   -o shairport-sync shairport.o rtsp.o mdns.o mdns_external.o common.o rtp.o player.o alac.o audio.o  mdns_avahi.o  audio_alsa.o   audio_pipe.o       -lasound  -lavahi-common -lavahi-client -lsoxr -lssl -lcrypto  -lconfig  -lpopt  -lm -lpthread -ldaemon  -lrt
make[2]: Leaving directory '/tmp/shairport-sync'
make[1]: Leaving directory '/tmp/shairport-sync'
Making install in man
make[1]: Entering directory '/tmp/shairport-sync/man'
make[2]: Entering directory '/tmp/shairport-sync/man'
make[2]: Nothing to be done for 'install-exec-am'.
 /bin/mkdir -p '/usr/local/share/man/man7'
 /usr/bin/install -c -m 644 shairport-sync.7 '/usr/local/share/man/man7'
make[2]: Leaving directory '/tmp/shairport-sync/man'
make[1]: Leaving directory '/tmp/shairport-sync/man'
make[1]: Entering directory '/tmp/shairport-sync'
make[2]: Entering directory '/tmp/shairport-sync'
 /bin/mkdir -p '/usr/local/bin'
  /usr/bin/install -c shairport-sync '/usr/local/bin'
make  install-exec-hook
make[3]: Entering directory '/tmp/shairport-sync'
[ -e /usr/local/etc ] || mkdir /usr/local/etc
cp scripts/shairport-sync.conf /usr/local/etc/shairport-sync.conf.sample
[ -f /usr/local/etc/shairport-sync.conf ] || cp scripts/shairport-sync.conf /usr/local/etc/shairport-sync.conf
[ -e /lib/systemd/system ] || mkdir -p /lib/systemd/system
[ -f /lib/systemd/system/shairport-sync.service ] || cp scripts/shairport-sync.service /lib/systemd/system
make[3]: Leaving directory '/tmp/shairport-sync'
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/tmp/shairport-sync'
make[1]: Leaving directory '/tmp/shairport-sync'

######################################################################################################
-- installing Assistant - this might take a while, go grab a coffee and check back in 10-15min --
-- in the meantime, go visit
https://developers.google.com/assistant/sdk/prototype/getting-started-pi-python/config-dev-project-and-account
-- Get your OAuth Credentials-JSON file ready --
-- and rename/move it to /home/pi/Downloads/client_secret.json --
######################################################################################################

Reading package lists... Done
Building dependency tree
Reading state information... Done
python3-dev is already the newest version.
python3-venv is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 9 not upgraded.
Reading package lists... Done
Building dependency tree
Reading state information... Done
libffi-dev is already the newest version.
libssl-dev is already the newest version.
portaudio19-dev is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 9 not upgraded.
Could not import runpy module
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 2237, in _find_and_load
  File "<frozen importlib._bootstrap>", line 2222, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 2164, in _find_spec
  File "<frozen importlib._bootstrap>", line 1940, in find_spec
  File "<frozen importlib._bootstrap>", line 1911, in _get_spec
  File "<frozen importlib._bootstrap>", line 1879, in _path_importer_cache
FileNotFoundError: [Errno 2] No such file or directory
pi@raspberrypi:~ $
`

Latest version of raspbian

xtools-at, 2017-05-07:
the error happens here, in the install_assistant function:
https://github.com/xtools-at/AssistantPi/blob/master/src/scripts/inc/os/debian.sh
when running `python3 -m venv /opt/AlexaPi/env`

did you clone the repository to `/opt/AlexaPi` ? please try and run `cd /opt; ls` and check if there's a folder called AlexaPi.

claudioita, 2017-05-07:
how to do this
?
did you clone the repository to /opt/AlexaPi ? please try and run cd /opt; ls and check if there's a folder called AlexaPi.

xtools-at, 2017-05-07:
I don't know what's causing this, installer get's stuck when creating the virtual environment for Google Assistant SDK (--> `python3 -m venv /opt/AlexaPi/env`).
Can you try run this command for it's own and see what it says/does? Please try running as Root:
`sudo python3 -m venv /opt/AlexaPi/env`

The only idea I've got is that you haven't installed Python 3 - but that shouldn't be the case, if you're using a fresh install of the actual version of Raspbian Jessie from April 2017.
You could try `sudo apt-get install python3` and see if something gets installed. If so, try running the setup again.

claudioita, 2017-05-07:
I did this (as user pi)

Could not import runpy module
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 2237, in _find_and_load
  File "<frozen importlib._bootstrap>", line 2222, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 2164, in _find_spec
  File "<frozen importlib._bootstrap>", line 1940, in find_spec
  File "<frozen importlib._bootstrap>", line 1911, in _get_spec
  File "<frozen importlib._bootstrap>", line 1879, in _path_importer_cache
FileNotFoundError: [Errno 2] No such file or directory
pi@raspberrypi:/opt/AlexaPi/src/scripts $ sudo python3 -m venv /opt/AlexaPi/env
pi@raspberrypi:/opt/AlexaPi/src/scripts $ sudo apt-get install python3
Reading package lists... Done
Building dependency tree
Reading state information... Done
python3 is already the newest version.
The following packages were automatically installed and are no longer required:
  python-chardet python-colorama python-distlib python-html5lib python-ndg-httpsclient python-pyasn1
  python-requests python-six python-urllib3 python-wheel
Use 'apt-get autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 57 not upgraded.
pi@raspberrypi:/opt/AlexaPi/src/scripts $

claudioita, 2017-05-07:
and as root

root@raspberrypi:/home/pi# sudo python3 -m venv /opt/AlexaPi/env
root@raspberrypi:/home/pi#

xtools-at, 2017-05-07:
just found [this](https://bugs.launchpad.net/ubuntu/+source/python3-defaults/+bug/1024016)
it seems to be a bug in an older python3 version. try running
`sudo apt-get remove python3`
`sudo apt-get install python3`
and then do the setup again.

Or simpler:
get a the **actual** [Raspbian Jesse Lite](https://www.raspberrypi.org/downloads/raspbian/), do a fresh install, then run the setup. this works perfectly fine for me, did this 3 times while verifying that the setup works and hadn't had any problems.

claudioita, 2017-05-07:
I guess the Google Assistant did not install so I did this
cd/opt
python -m pip install google-assistant-sdk[samples]

hope is not too wrong

xtools-at, 2017-05-07:
This won't work with AssistantPi, only for it's own (i.e. running Assistant from command line).
If you want to use AssistantPi, please try the suggestions I've already given.

claudioita, 2017-05-07:
I did, 
sudo apt-get remove python3
sudo apt-get install python3
and then do the setup again

I receive same error as before

xtools-at, 2017-05-07:
and you didn't try setting it up on a fresh and actual Raspbian install. I can't help you any further if you don't try this, as I can't reproduce your issue.

claudioita, 2017-05-07:
tI am unable to install a jessie lite (I am headless), since I can't find the pi3 on my network.
I give up.

Thanks anyway

MarMed, 2017-05-07:
My issue was opened having used a new rasbian lite image.

xtools-at, 2017-05-07:
@MarMed did you use the latest version of April? Or did you do an `apt-get upgrade` before?

If so, please try the following (removes AssistantPi, clones it again, sets up virtual environment for Assistant SDK):
```
sudo rm -rf /opt/AlexaPi
cd /opt
sudo git clone https://github.com/xtools-at/AssistantPi.git AlexaPi
sudo python3 -m venv /opt/AlexaPi/env
```
make sure there's a folder called **env** in `/opt/AlexaPi` (that's where the installer crashes) and then run the setup again.

beeme1mr, 2017-05-08:
I ran into the same issue but simply running the script worked.  I'm now stuck on the the OAuth process for the Google Assistant.  I'll open an issue with more information if I'm unable to get it working.

claudioita, 2017-05-08:
I followed the instructions, still errors, this time the env directory is present (before that was definetely an issue, the env directory was missing)

######################################################################################################
-- installing Assistant - this might take a while, go grab a coffee and check back in 10-15min --
-- in the meantime, go visit
https://developers.google.com/assistant/sdk/prototype/getting-started-pi-python/config-dev-project-and-account
-- Get your OAuth Credentials-JSON file ready --
-- and rename/move it to /home/pi/Downloads/client_secret.json --
######################################################################################################

Reading package lists... Done
Building dependency tree
Reading state information... Done
python3 is already the newest version.
python3-dev is already the newest version.
python3-venv is already the newest version.
The following packages were automatically installed and are no longer required:
  docutils-common gir1.2-gmenu-3.0 libgnome-menu-3-0 libpackagekit-glib2-18
  librtimulib-dev librtimulib-utils librtimulib7 pi-package-data
  python-chardet python-colorama python-distlib python-html5lib
  python-ndg-httpsclient python-pyasn1 python-requests python-rtimulib
  python-sense-hat python-six python-urllib3 python-wheel python3-rtimulib
  sphinx-common
Use 'apt-get autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
Reading package lists... Done
Building dependency tree
Reading state information... Done
libffi-dev is already the newest version.
libssl-dev is already the newest version.
portaudio19-dev is already the newest version.
The following packages were automatically installed and are no longer required:
  docutils-common gir1.2-gmenu-3.0 libgnome-menu-3-0 libpackagekit-glib2-18
  librtimulib-dev librtimulib-utils librtimulib7 pi-package-data
  python-chardet python-colorama python-distlib python-html5lib
  python-ndg-httpsclient python-pyasn1 python-requests python-rtimulib
  python-sense-hat python-six python-urllib3 python-wheel python3-rtimulib
  sphinx-common
Use 'apt-get autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
Could not import runpy module
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 2237, in _find_and_load
  File "<frozen importlib._bootstrap>", line 2222, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 2164, in _find_spec
  File "<frozen importlib._bootstrap>", line 1940, in find_spec
  File "<frozen importlib._bootstrap>", line 1911, in _get_spec
  File "<frozen importlib._bootstrap>", line 1879, in _path_importer_cache
FileNotFoundError: [Errno 2] No such file or directory
pi@raspberrypi:/opt $

dpoulter2, 2017-05-08:
I am also having the same issue. Have tried all your suggestions. No success. Was a brand new install of Raspbian Jessie Lite

jefforion, 2017-05-08:
Hi,
I have a same issues.
My Raspbian version it's a fresh version and the last "RASPBIAN JESSIE WITH PIXEL" (Version: April 2017)

I don't have a folder name "env" after the first run.
As xtools-at said, I did one : `sudo python3 -m venv /opt/AlexaPi/env`
It's same error.

Then I uninstall and reinstall Pyth.
It's same error.

It's same with / without Airplay.
Without Google Assistant it works, juste, I don't have Google Assitant :(

xtools-at, 2017-05-08:
Hi guys,
ok, this might be a clue. I've used **Raspbian Jessie Lite** all along, maybe it's a bug related to having Pixel installed.

I'll look into that, in the meantime this could help you out:
- In the Setup, don't install Google Assistant, but finish everything AlexaPi-related. The setup shouldn't have crashed now and the AlexaPi should be running.
- Run `python3 -m venv /opt/AlexaPi/env` and check if there is a folder `env` in _/opt/AlexaPi_
- Run the steps of AssistantPi's Google Assistant setup manually:
```
sudo /opt/AlexaPi/env/bin/pip install pip setuptools --upgrade

cd /opt/AlexaPi/src
sudo rm -rf assistant-sdk-python
sudo git clone https://github.com/xtools-at/assistant-sdk-python.git
cd assistant-sdk-python
sudo /opt/AlexaPi/env/bin/python -m pip install --upgrade -e ".[samples]"

cp /opt/AlexaPi/src/assistant.example.asoundrc /home/pi/.asoundrc
```
- then, run Authentication with Google manually:
```
sudo /opt/AlexaPi/env/bin/python -m googlesamples.assistant.auth_helpers --client-secrets /home/pi/Downloads/client_secret.json
```

Hope this helps!

dpoulter2, 2017-05-08:
I have got the version without Pixel and it still isn't working. I will try your solution and see if it fixes the problem

DocCheese, 2017-05-09:
In opt type "sudo mv AlexiaPi AlexaPi". 

Try that.

DocCheese, 2017-05-09:
I'm having the same issue with Pixel and lite variants.

xtools-at, 2017-05-09:
--> added the manual installation instructions to README, still haven't figured out the problem

samiultahir, 2017-05-09:
I have the same issue here!
A fix would be nice :)

xtools-at, 2017-05-10:
wasn't able to replicate it yet, please use the manual workaround meanwhile

claudioita, 2017-05-10:
with the workaround I could install Google Assistant

goodpunk6, 2017-05-11:
Have the same issue. will try the workaround.

goodpunk6, 2017-05-12:
I tried the workaround and everything seems to have installed. I can record and playback my recordings. However, when i say alexa or google, nothing happens.

xtools-at, 2017-05-13:
Hi there,

I've re-written a big part of AssistantPi, this makes Assistant a lot snappier and should fix the setup problem too.

Please run
```
cd /opt/AlexaPi
sudo git pull
sudo /opt/AlexaPi/src/scripts/setup.sh
```
and do the setup again, including Auth with Google and Amazon.

closed 2017-05-13 by xtools-at

goodpunk6, 2017-05-13:
just recreated the sd card with Jessie Lite like before. installed the
latest version of assistant pi. was never prompted to authenticate with
google. Do I do that manually?

goodpunk6, 2017-05-13:
Ok so I reran everything from the top and I'm pretty sure nothing google
related installed.

When i attempt to verify the audio setup

pi@raspberrypi:~ $ python -m googlesamples.assistant.audio_helpers
/usr/bin/python: No module named googlesamples.assistant
pi@raspberrypi:~ $

that is what I get.