xtools-at/AssistantPi #2: error runnung setup.sh

claudioita, 2017-05-07:
**Your OS (including version) where you are running AlexaPi:**
```raspberry hassbian
```

**Your hardware platform and model you are running on:**
```pi3

```

**Python release (`python2 --version`):**
```3

```

**Description of problem:**
```setup.sh gives error. I can use ALexaPi and I can use Gooogle Home. Each separately. I do have a mic/speaker on USB

```

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```The following packages were automatically installed and are no longer required:
  python-chardet python-colorama python-distlib python-html5lib python-ndg-httpsclient python-pyasn1
  python-requests python-six python-urllib3 python-wheel
Use 'apt-get autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 57 not upgraded.
Could not import runpy module
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 2237, in _find_and_load
  File "<frozen importlib._bootstrap>", line 2222, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 2164, in _find_spec
  File "<frozen importlib._bootstrap>", line 1940, in find_spec
  File "<frozen importlib._bootstrap>", line 1911, in _get_spec
  File "<frozen importlib._bootstrap>", line 1879, in _path_importer_cache
FileNotFoundError: [Errno 2] No such file or directory
pi@raspberrypi:/opt $ ^C
pi@raspberrypi:/opt $ /opt/AlexaPi/env/bin/python -m googlesamples.assistant.auth_helpers --client-secrets /home/pi/Downloads/client_secret.json
-bash: /opt/AlexaPi/env/bin/python: No such file or directory
pi@raspberrypi:/opt $ python /opt/AlexaPi/src/main.py
2017-05-07 18:32:01 CRITICAL: Your input_device 'plughw:CARD=Device,DEV=0' is invalid. Use one of the following:
null
pulse
default
sysdefault:CARD=USB
front:CARD=USB,DEV=0
surround21:CARD=USB,DEV=0
surround40:CARD=USB,DEV=0
surround41:CARD=USB,DEV=0
surround50:CARD=USB,DEV=0
surround51:CARD=USB,DEV=0
surround71:CARD=USB,DEV=0
iec958:CARD=USB,DEV=0
dmix:CARD=USB,DEV=0
dsnoop:CARD=USB,DEV=0
hw:CARD=USB,DEV=0
plughw:CARD=USB,DEV=0
pi@raspberrypi:/opt $

```

xtools-at, 2017-05-07:
Hi there,

the setup didn't finish, so you can't run main.py yet.

I don't know what's causing this, installer get's stuck when creating the virtual environment for Google Assistant SDK (--> `python3 -m venv /opt/AlexaPi/env`).
Can you try run this command for it's own and see what it says/does? Please try running as Root:
`sudo python3 -m venv /opt/AlexaPi/env`

The only idea I've got is that you haven't installed Python 3 - but that shouldn't be the case, if you're using a fresh install of the actual version of Raspbian Jessie from April 2017.
You could try `sudo apt-get install python3` and see if something gets installed. If so, try running the setup again.

xtools-at, 2017-05-07:
I'm closing this, as a duplicate of Issue #1

closed 2017-05-07 by xtools-at