xtools-at/AssistantPi #3: Google does not answer a question, Alexa does.

TinajaLabs, 2017-05-07:
Just testing that I can get the time back from Alexa and Google triggers:  Alexa works, Google does not.  "Google: triggers and responds with a "yes", but when asked for a time it does not respond.

**Your OS (including version) where you are running AlexaPi:**
```
raspbian jessie latest upgraded
```

**Your hardware platform and model you are running on:**
```
raspi 3
```

**Python release (`python2 --version`):**
```
Python 2.7.9
```

**Description of problem:**
```

```

**Steps to reproduce:**

1. "google"
2. "yes"
3. "what time is it"
4. can hear an audio tinkle sound...  then nothing

**Traceback (if applicable):**
from running AlexaPi running non-service in debug mode:
```
sudo systemctl stop AlexaPi.service
/opt/AlexaPi/src/main.py -d
```

```
017-05-07 14:30:55 DEBUG: Playing audio: /opt/AlexaPi/src/resources/alexayes.mp3
2017-05-07 14:30:55 DEBUG: Started play. sox -q /opt/AlexaPi/src/resources/alexayes.mp3 -t alsa default vol -6 dB pad 0 0
2017-05-07 14:30:56 DEBUG: Finished play. 
2017-05-07 14:30:56 DEBUG: Pocketsphinx Trigger found voice command: **google**
2017-05-07 14:30:56 DEBUG: Compare to Assistant Command from Config: **google**
2017-05-07 14:30:56 DEBUG: Assistant triggered, starting...
sox WARN mp3-util: MAD lost sync
INFO:root:Connecting to embeddedassistant.googleapis.com
INFO:root:Recording audio request.
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
INFO:root:End of audio request detected
INFO:root:Finished playing assistant response.
INFO:root:Assistant conversation finished.
2017-05-07 14:31:10 DEBUG: assistant_handler: conversation complete, moving on
```
Any tips appreciated,
Chris.

xtools-at, 2017-05-07:
Hi Chris,

Assistant needs a second or two to connect to the Google servers.
The "yes" you're hearing is Alexa's answer to the hotword and actually a bug which I fixed today (just do a `cd /opt/AlexaPi; sudo git pull` to update). After updating, you'll only hear this when you call Alexa.

The "tinkle sound" is actually the sign that the connection to Google API has been established and you can start speaking now - try speaking after the beep :)

Bests,
Martin

TinajaLabs, 2017-05-07:
Thank you.  I just now came to post my conclusion that it seems to be a timing issue.  You beat me to it.  The timing is a bit awkward.  The way it seems is that it goes from say yes, then after 4-5 seconds I hear the google listening sound, the INFO:root:Recording audio request shows up a few seconds later and often it misses my full statement/question.

I'll update as you suggest and see how that goes.

Thanks so much for your help.

```
2017-05-07 15:50:06 DEBUG: Playing audio: /opt/AlexaPi/src/resources/alexayes.mp3
2017-05-07 15:50:06 DEBUG: Started play. sox -q /opt/AlexaPi/src/resources/alexayes.mp3 -t alsa default vol -6 dB pad 0 0
2017-05-07 15:50:07 DEBUG: Finished play. 
2017-05-07 15:50:07 DEBUG: Pocketsphinx Trigger found voice command: **google**
2017-05-07 15:50:07 DEBUG: Compare to Assistant Command from Config: **google**
2017-05-07 15:50:07 DEBUG: Assistant triggered, starting...
sox WARN mp3-util: MAD lost sync
INFO:root:Connecting to embeddedassistant.googleapis.com
INFO:root:Recording audio request.
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
INFO:root:End of audio request detected
INFO:root:Transcript of user request: "is it".
INFO:root:Playing assistant response.
INFO:root:Finished playing assistant response.
INFO:root:Assistant conversation finished.
2017-05-07 15:50:47 DEBUG: assistant_handler: conversation complete, moving on
```

xtools-at, 2017-05-07:
I'll try to establish the connection to Google beforehand, to make the response quicker, but this needs some effort.

TinajaLabs, 2017-05-08:
Leaving some feedback:

The trigger, "google"  no longer says, "yes".  After a few seconds it goes directly to the google waiting sound.  Then I wait another few seconds to ask my question.  Sometimes I get a hit and when I do, the google playback is jumpy (perhaps an audio playback problem).

Below, is from a successful response of "google" : "what time is it", but you can see I repeated the word "google" as I didn't think it had caught the trigger.

I suspect the biggest problem with using raspis with microphones and speakers is to find a consistent cadence in the speech pattern that always works; even if it's a little slow, the cadence needs to be consistent. 
```
2017-05-07 16:50:05 DEBUG: Pocketsphinx Trigger found voice command: **google**
2017-05-07 16:50:05 DEBUG: Compare to Assistant Command from Config: **google**
2017-05-07 16:50:05 DEBUG: Assistant triggered, starting...
sox WARN mp3-util: MAD lost sync
INFO:root:Connecting to embeddedassistant.googleapis.com
INFO:root:Recording audio request.
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
INFO:root:End of audio request detected
INFO:root:Transcript of user request: "Google".
INFO:root:Playing assistant response.
INFO:root:Expecting follow-on query from user.
WARNING:root:SoundDeviceStream write underflow (size: 1600)
INFO:root:Finished playing assistant response.
INFO:root:Assistant conversation continues.
INFO:root:Recording audio request.
INFO:root:End of audio request detected
INFO:root:Transcript of user request: "what time is it".
INFO:root:Playing assistant response.
WARNING:root:SoundDeviceStream write underflow (size: 1600)
WARNING:root:SoundDeviceStream write underflow (size: 1600)
INFO:root:Finished playing assistant response.
INFO:root:Assistant conversation finished.
2017-05-07 16:50:40 DEBUG: assistant_handler: conversation complete, moving on
```

Thanks,
Chris.

xtools-at, 2017-05-08:
Hi Chris,
thanks for the Feedback! I tried to come up with the real solution, but it's still buggy. I'll need more time to figure that out.
In the meantime, I've remedied it a bit. It still takes a few seconds for Google to come up, but the Start-Sound is now in sync with the actual start of the recording.

Needs some updating though of both AssistantPi and the SDK, this should do the trick:
```
cd /opt/AlexaPi
sudo git pull

cd /opt/AlexaPi/src/assistant-sdk-python
sudo git pull
sudo /opt/AlexaPi/env/bin/python -m pip install --upgrade -e ".[samples]"
```

The choppy audio playback is an issue with the SDK, but you can find information here on how to fine tune that:
https://developers.google.com/assistant/sdk/prototype/getting-started-pi-python/troubleshooting
(-> Audio Issues -> Third Bulletpoint)

xtools-at, 2017-05-09:
--> Renamed issue from "Google does not answer a question, Alexa does."

xtools-at, 2017-05-09:
Hi Chris,

I've added the possibility to pass the block- and flush-rates into Assistant SDK. Do a pull on the main repo and then in `/etc/opt/AlexaPi/config.yaml`
- find `sound` attribute
- add sub-attributes and your values:
```
  assistant:
    block_size: ""
    flush_size: ""
```
( like in this [diff](https://github.com/xtools-at/AssistantPi/commit/2bee166280702985987c6f39a156217ff01eb104#diff-b034bbe11dab609f5df289877a107281) )

xtools-at, 2017-05-13:
Hi there,

I've re-written a big part of AssistantPi, Assistant now responds almost immediately. For the choppy audio, I'll push an update tomorrow which is still in testing.

Please run
```
cd /opt/AlexaPi
sudo git pull
sudo /opt/AlexaPi/src/scripts/setup.sh
```
and do the setup again, including Auth with Google and Amazon.

closed 2017-05-13 by xtools-at