xtools-at/AssistantPi #4: ALexa is fast, Google takes minutes to "initialize"

claudioita, 2017-05-08:
"Hello" is spoken almost immediately

The blink sound (I think of initializing Google) it takes long, like few minutes

claudioita, 2017-05-09:
is the above normal behaviour?

xtools-at, 2017-05-09:
> is the above normal behaviour?
no it's not, seems that the Google-sound is blocking the audio resources for the Assistant SDK. I'll have a look into this on the weekend.

claudioita, 2017-05-09:
it should say "Hello" (Alexa) and blink sound (Google Assistant) almost together, correct?

xtools-at, 2017-05-09:
no, the _hello_ is just that the sign that AlexPi is initialized.
the Google-sound plays when the Google-hotword has been recognized, the connection to Google servers is established and it starts recording

xtools-at, 2017-05-13:
Hi there,
I've re-written a big part of AssistantPi, this shouldn't be an issue anymore. Please re-open the ticket if you're still experiencing this problem.

Please run
```
cd /opt/AlexaPi
sudo git pull
sudo /opt/AlexaPi/src/scripts/setup.sh
```
and do the setup again, including Auth with Google and Amazon.

closed 2017-05-13 by xtools-at