xtools-at/AssistantPi #5: Adding AssistantPi to "MagPi Release" (using Google Voice Hat, etc)

ShaneC80, 2017-05-08:
Attempting to configure AssistantPi to work with the MagPi Google Assistant Release

Alexa and Voice Recognizer both inoperable after install.
=================================================
Hardware:  Pi3B
=================================================
"Google AIY Projects Voice Kit SD Image" - configured and working initially, prior to Assistant Pi Install
Pi-Hole
Python 2.7.9 & Python 3.4.2 versions.
=================================================
Google Assistant already installed from google image.
=================================================
 
Ran setup script, installed as system md, no air play support, when prompted for Google Assistant config, chose "No".   Setup appears to complete successfully.

Alexa and Voice Recognizer both inoperable after install and after reboot.
=================================================

arecord -l && aplay -l

pi@PiVoice:/opt/AlexaPi/src/scripts $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-hifi-0 []
  Subdevices: 0/1
  Subdevice #0: subdevice #0
pi@PiVoice:/opt/AlexaPi/src/scripts $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

============================================

systemctl output:

* AlexaPi.service                 loaded failed failed    Alexa client for all your devices

pi@PiVoice:~ $ systemctl status AlexaPi.service
* AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
   Active: failed (Result: exit-code) since Mon 2017-05-08 12:50:53 EDT; 14min ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
  Process: 625 ExecStart=/usr/bin/python /opt/AlexaPi/src/main.py --daemon (code=exited, status=1/FAILURE)
 Main PID: 625 (code=exited, status=1/FAILURE)
==========================================================
pi@PiVoice:~ $ systemctl status voice-recognizer.service
* voice-recognizer.service - voice recognizer
   Loaded: loaded (/lib/systemd/system/voice-recognizer.service; enabled)
   Active: active (running) since Mon 2017-05-08 13:08:11 EDT; 574ms ago
 Main PID: 4143 (python3)
   CGroup: /system.slice/voice-recognizer.service
           `-4143 /home/pi/voice-recognizer-raspi/env/bin/python3 -u src/main.py
============================================================

xtools-at, 2017-05-09:
Hi,

before I'll look into this, just a quick suggestion:

You wrote `Google Assistant already installed from google image.` - AssistantPi uses it's own fork of the Assistant SDK in it's own virtual environment and is dependent on having that. It won't work this way, you have run the setup and install Google Assistant from there too.
Please perform this step and let me know if the problem persits.

Bests, Martin

ShaneC80, 2017-05-09:
Martin,

That would be one big reason it's not working then.  Let me do some more playing around and see if I can get some new info.

Is there a way to keep this thread open but "not a support issue" for experimenting and data gathering on the PiHat and various other shenanigans?

xtools-at, 2017-05-09:
yes of course, tell me how it's going :)

ShaneC80, 2017-05-09:
Brought everything back up with just the Google AIY running from the MagPie kit, then ran the AssistantPi setup.  

Here's my Alexa logs.

-- Logs begin at Tue 2017-05-09 12:08:56 EDT, end at Tue 2017-05-09 12:52:14 EDT. --
May 09 12:49:28 PiVoice systemd[1]: Starting Alexa client for all your devices...
May 09 12:49:28 PiVoice systemd[1]: Started Alexa client for all your devices.
May 09 12:49:29 PiVoice pulseaudio[1579]: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-da
May 09 12:49:29 PiVoice pulseaudio[1579]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon with
May 09 12:49:29 PiVoice pulseaudio[1579]: [pulseaudio] bluez4-util.c: org.bluez.Manager.GetProperties() failed: org.freedesktop.DBus.Error.UnknownMethod: Method "GetPro
May 09 12:49:29 PiVoice python[1566]: CRITICAL: Your input_device 'plughw:CARD=Device,DEV=0' is invalid. Use one of the following:
May 09 12:49:29 PiVoice python[1566]: null
May 09 12:49:29 PiVoice python[1566]: pulse
May 09 12:49:29 PiVoice python[1566]: softvol
May 09 12:49:29 PiVoice python[1566]: micboost
May 09 12:49:29 PiVoice python[1566]: default
May 09 12:49:29 PiVoice python[1566]: sysdefault:CARD=sndrpigooglevoi
May 09 12:49:29 PiVoice python[1566]: dmix:CARD=sndrpigooglevoi,DEV=0
May 09 12:49:29 PiVoice python[1566]: dsnoop:CARD=sndrpigooglevoi,DEV=0
May 09 12:49:29 PiVoice python[1566]: hw:CARD=sndrpigooglevoi,DEV=0
May 09 12:49:29 PiVoice python[1566]: plughw:CARD=sndrpigooglevoi,DEV=0
May 09 12:49:29 PiVoice systemd[1]: AlexaPi.service: main process exited, code=exited, status=1/FAILURE
May 09 12:49:29 PiVoice systemd[1]: Unit AlexaPi.service entered failed state.
May 09 12:51:01 PiVoice systemd[1]: Stopped Alexa client for all your devices.
May 09 12:51:05 PiVoice systemd[1]: Starting Alexa client for all your devices...
May 09 12:51:05 PiVoice systemd[1]: Started Alexa client for all your devices.
May 09 12:51:06 PiVoice pulseaudio[1633]: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-da
May 09 12:51:06 PiVoice pulseaudio[1633]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon with
May 09 12:51:06 PiVoice pulseaudio[1633]: [pulseaudio] bluez4-util.c: org.bluez.Manager.GetProperties() failed: org.freedesktop.DBus.Error.UnknownMethod: Method "GetPro
May 09 12:51:06 PiVoice python[1620]: CRITICAL: Your input_device 'plughw:CARD=Device,DEV=0' is invalid. Use one of the following:
May 09 12:51:06 PiVoice python[1620]: null
May 09 12:51:06 PiVoice python[1620]: pulse
May 09 12:51:06 PiVoice python[1620]: softvol
May 09 12:51:06 PiVoice python[1620]: micboost
May 09 12:51:06 PiVoice python[1620]: default
May 09 12:51:06 PiVoice python[1620]: sysdefault:CARD=sndrpigooglevoi
May 09 12:51:06 PiVoice python[1620]: dmix:CARD=sndrpigooglevoi,DEV=0
May 09 12:51:06 PiVoice python[1620]: dsnoop:CARD=sndrpigooglevoi,DEV=0
May 09 12:51:06 PiVoice python[1620]: hw:CARD=sndrpigooglevoi,DEV=0
May 09 12:51:06 PiVoice python[1620]: plughw:CARD=sndrpigooglevoi,DEV=0
May 09 12:51:06 PiVoice systemd[1]: AlexaPi.service: main process exited, code=exited, status=1/FAILURE
May 09 12:51:06 PiVoice systemd[1]: Unit AlexaPi.service entered failed state.
May 09 12:51:11 PiVoice systemd[1]: Stopped Alexa client for all your devices.
May 09 12:52:02 PiVoice systemd[1]: Stopped Alexa client for all your devices.

ShaneC80, 2017-05-09:
Hmm....maybe getting closer:

pi@PiVoice:/opt/AlexaPi/src/scripts $ sudo systemctl status AlexaPi.service
* AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
   Active: failed (Result: exit-code) since Tue 2017-05-09 16:22:38 EDT; 25s ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
  Process: 2767 ExecStart=/usr/bin/python /opt/AlexaPi/src/main.py --daemon (code=exited, status=1/FAILURE)
 Main PID: 2767 (code=exited, status=1/FAILURE)

May 09 16:22:38 PiVoice python[2767]: softvol
May 09 16:22:38 PiVoice python[2767]: micboost
May 09 16:22:38 PiVoice python[2767]: default
May 09 16:22:38 PiVoice python[2767]: sysdefault:CARD=sndrpigooglevoi
May 09 16:22:38 PiVoice python[2767]: dmix:CARD=sndrpigooglevoi,DEV=0
May 09 16:22:38 PiVoice python[2767]: dsnoop:CARD=sndrpigooglevoi,DEV=0
May 09 16:22:38 PiVoice python[2767]: hw:CARD=sndrpigooglevoi,DEV=0
May 09 16:22:38 PiVoice python[2767]: plughw:CARD=sndrpigooglevoi,DEV=0
May 09 16:22:38 PiVoice systemd[1]: AlexaPi.service: main process exited, code=exited, status=1/FAILURE
May 09 16:22:38 PiVoice systemd[1]: Unit AlexaPi.service entered failed state.

xtools-at, 2017-05-10:
this logs look like you'd be running pulseaudio, maybe through a Desktop GUI or the Google AIY image.
AlexaPi has it's problems with that, please make sure to check this guide where it says "PulseAudio":
https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging

the error message below points to AlexaPi config. please open `/etc/opt/AlexaPi/config.yaml`, find the attribute `input device` and overwrite the value with the one of those given by the output message. I think `plughw:CARD=sndrpigooglevoi,DEV=0` will work.

following [this guide](https://developers.google.com/assistant/sdk/prototype/getting-started-pi-python/configure-audio) and your arecord/aplay output, you might have to adjust your `/home/pi/.asoundrc` as well. both mic and speaker should have `pcm "hw:0,0"` as both your audio capture and output are "card:0 device:0"

make sure to stop and restart AlexaPi service or reboot after all of that.

ShaneC80, 2017-05-10:
Disabled "voice-recognition.service" as it was cooking the processor looping starts and 
stops.  Additionally this way the only thing we're troubleshooting is AlexaPi (mostly).

>this logs look like you'd be running pulseaudio, maybe through a Desktop GUI or the Google AIY >image.  >AlexaPi has it's problems with that, please make sure to check this guide where it says >"PulseAudio": https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging

The Google AIY Image has the GUI.  I installed PulseAudio Manager to try to do something with Pulse, it's unable to connect.  Attempted the disabling of Pulse as outlined in the AlexaPi link.

Under the GUI "speaker" icon, the sound card is ID'd as "snd_rpi_googlevoicehat_soundar (Alsa mixer)(Default)

"arecord" and "aplayback" are both functional with "plughw:CARD=sndrpigooglevoi,DEV=0", so that's what's configured in /etc/opt/AlexaPi/config.yaml

>following this guide and your arecord/aplay output, you might have to adjust your /home/pi/.asoundrc as well. both mic and speaker should have pcm "hw:0,0" as both your audio capture and output are "card:0 device:0"

usb_mic is defined with pcm "hw:1,0" - I'm wondering if that's because of the Google Hat having the mic as a daughter board?

Latest issue is getting a refresh token:

May 10 07:53:32 PiVoice systemd[1]: Starting Alexa client for all your devices...
May 10 07:53:32 PiVoice systemd[1]: Started Alexa client for all your devices.
May 10 07:53:33 PiVoice pulseaudio[1736]: [pulseaudio] sink.c: Default and alternate sample rates are the same.
May 10 07:53:33 PiVoice pulseaudio[1736]: [pulseaudio] source.c: Default and alternate sample rates are the same.
May 10 07:53:33 PiVoice pulseaudio[1736]: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-da
May 10 07:53:33 PiVoice pulseaudio[1736]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon with
May 10 07:53:33 PiVoice python[1721]: CRITICAL: Your input_device 'plughw:CARD=Device,DEV=0' is invalid. Use one of the following:

xtools-at, 2017-05-11:
Hi,
the logs are telling me it's not about a refresh token :)

pulseaudio is still having troubles, I've just ran into the same issue. I did the step (**Running PA in system-wide mode**)[https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging], then it finally worked.

and your changes to the `/etc/opt/AlexaPi/config.yaml` might not have been saved, as AlexaPi ist still complaining about having _plughw:CARD=Device,DEV=0_ as _input_device_

Bests,
Martin

xtools-at, 2017-05-13:
Hi there,
I've re-written a big part of AssistantPi, it now (among other things) utilizes Pulseaudio by default (as running as daemon was dependent on that), the setup does all the necessary settings.
Please run
```
cd /opt/AlexaPi
sudo git pull
sudo /opt/AlexaPi/src/scripts/setup.sh
```
and do the setup again, including Auth with Google and Amazon.

This should fix your issues.

leepark, 2017-05-15:
I've followed the instructions as best i could, started off with the AIY image and have used the latest AssistantPi git code. AlexaPi and Assistant seem to be running fine, here's the logs. but I have no Mic or Speakers working. either usb mic and 3.5mm audio or the google sound card hat. Any ideas? my config file is still using pulse as input and output

`pi@raspberrypi:~ $ sudo journalctl -u AlexaPi.service
-- Logs begin at Tue 2017-05-23 22:24:30 BST, end at Mon 2017-05-15 23:29:34 BST
May 23 22:24:34 raspberrypi systemd[1]: Starting Alexa client for all your devic
May 23 22:24:34 raspberrypi systemd[1]: Started Alexa client for all your device
May 15 23:21:57 raspberrypi python[605]: INFO: Checking Internet Connection ...
May 15 23:21:58 raspberrypi python[605]: INFO: Connection OK
May 15 23:21:58 raspberrypi python[605]: INFO: AVS token: Requesting a new one
May 15 23:21:59 raspberrypi python[605]: INFO: AVS token: Obtained successfully
pi@raspberrypi:~ $ sudo journalctl -u voice-recognizer -n 10 -f
-- Logs begin at Tue 2017-05-23 22:24:30 BST. --
May 23 22:24:38 raspberrypi python3[599]: /home/pi/voice-recognizer-raspi/env/li                                                                                        b/python3.4/site-packages/urllib3/contrib/socks.py:37: DependencyWarning: SOCKS                                                                                         support in urllib3 requires the installation of optional dependencies: specifica                                                                                        lly, PySocks.  For more information, see https://urllib3.readthedocs.io/en/lates                                                                                        t/contrib.html#socks-proxies
May 23 22:24:38 raspberrypi python3[599]: DependencyWarning
May 15 23:21:58 raspberrypi python3[599]: [2017-05-15 23:21:58,052] INFO:audio:s                                                                                        tarted recording
May 15 23:21:58 raspberrypi python3[599]: [2017-05-15 23:21:58,083] INFO:main:re                                                                                        ady...
May 15 23:27:21 raspberrypi python3[599]: [2017-05-15 23:27:21,110] INFO:main:li                                                                                        stening...
May 15 23:27:21 raspberrypi python3[599]: [2017-05-15 23:27:21,113] INFO:main:re                                                                                        cognizing...
May 15 23:27:29 raspberrypi python3[599]: [2017-05-15 23:27:29,229] INFO:speech:                                                                                        event_type: 1
May 15 23:27:29 raspberrypi python3[599]: [2017-05-15 23:27:29,233] INFO:main:th                                                                                        inking...
May 15 23:27:29 raspberrypi python3[599]: [2017-05-15 23:27:29,292] WARNING:main                                                                                        :no command recognized
May 15 23:27:31 raspberrypi python3[599]: [2017-05-15 23:27:31,579] INFO:main:re                                                                                        ady...
May 15 23:31:31 raspberrypi python3[599]: [2017-05-15 23:31:31,066] INFO:main:listening...
May 15 23:31:31 raspberrypi python3[599]: [2017-05-15 23:31:31,068] INFO:main:recognizing...
May 15 23:31:39 raspberrypi python3[599]: [2017-05-15 23:31:39,236] INFO:speech:event_type: 1
May 15 23:31:39 raspberrypi python3[599]: [2017-05-15 23:31:39,238] INFO:main:thinking...
May 15 23:31:39 raspberrypi python3[599]: [2017-05-15 23:31:39,304] WARNING:main:no command recognized
May 15 23:31:41 raspberrypi python3[599]: [2017-05-15 23:31:41,542] INFO:main:ready...
`

leepark, 2017-05-15:
and here's my aplay and arecord outputs.
`pi@raspberrypi:~ $ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
pulse
    PulseAudio Sound Server
softvol
micboost
default
    Playback/recording through the PulseAudio sound server
mic
speaker
sysdefault:CARD=sndrpigooglevoi
    snd_rpi_googlevoicehat_soundcar,
    Default Audio Device
dmix:CARD=sndrpigooglevoi,DEV=0
    snd_rpi_googlevoicehat_soundcar,
    Direct sample mixing device
dsnoop:CARD=sndrpigooglevoi,DEV=0
    snd_rpi_googlevoicehat_soundcar,
    Direct sample snooping device
hw:CARD=sndrpigooglevoi,DEV=0
    snd_rpi_googlevoicehat_soundcar,
    Direct hardware device without any conversions
plughw:CARD=sndrpigooglevoi,DEV=0
    snd_rpi_googlevoicehat_soundcar,
    Hardware device with all software conversions

arecord

`pi@raspberrypi:~ $ arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
pulse
    PulseAudio Sound Server
softvol
micboost
default
    Playback/recording through the PulseAudio sound server
mic
speaker
sysdefault:CARD=sndrpigooglevoi
    snd_rpi_googlevoicehat_soundcar,
    Default Audio Device
dmix:CARD=sndrpigooglevoi,DEV=0
    snd_rpi_googlevoicehat_soundcar,
    Direct sample mixing device
dsnoop:CARD=sndrpigooglevoi,DEV=0
    snd_rpi_googlevoicehat_soundcar,
    Direct sample snooping device
hw:CARD=sndrpigooglevoi,DEV=0
    snd_rpi_googlevoicehat_soundcar,
    Direct hardware device without any conversions
plughw:CARD=sndrpigooglevoi,DEV=0
    snd_rpi_googlevoicehat_soundcar,
    Hardware device with all software conversions
`

xtools-at, 2017-05-16:
can't see any obvious issue in the logs - maybe it's simply not a good idea to install this on top of a system which is meant to do something else...

is AssistantPi even running? can't see that anywhere. please check
sudo systemctl status AlexaPi.service -l

leepark, 2017-05-16:
> can't see any obvious issue in the logs - maybe it's simply not a good idea to install this on top of a system which is meant to do something else...

> is AssistantPi even running? can't see that anywhere. please check
sudo systemctl status AlexaPi.service -l

Yeah, probably, I've had the AIY kit working and AlexaPi(with usb mic and speaker), but thought this was how ShaneC80 had done his, so wanted to replicate and have it all in a nice box(well cardboard!). I'll start fresh again tonight with a clean copy of Jessie lite.

xtools-at, 2017-05-16:
I think you can use the HAT coming with AIY, pulseaudio should pick that up - just don't use the image provided with the kit :)

leepark, 2017-05-16:
Pulse didn't work on a clean install, Alexa is up and running using usb mic and speaker but Google isnt, if i stop AlexaPi and run the command 
_source /opt/AlexaPi/env/bin/activate
### python -m googlesamples.assistant --credentials /etc/opt/AlexaPi/assistant_credentials.json
and type assistant_record into the prompt_
it works

logs
google sound card isn't detected now.
_pi@PiAssistant:~ $ arecord -l && aplay -l
**** List of CAPTURE Hardware Devices ****
card 1: U012250289 [USBMIC Serial# 012250289], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0_

the reason Assistant isnt running.

_pi@PiAssistant:~ $ systemctl status voice-recognizer.service
● voice-recognizer.service
   Loaded: not-found (Reason: No such file or directory)
   Active: inactive (dead)_

xtools-at, 2017-05-17:
there's no **voice-recognizer.service** coming with AssistantPi, that's Google AIY specific.

we're using the [Google Assistant Python SDK](https://github.com/googlesamples/assistant-sdk-python), which doesn't come with that service - the voice recognition is baked into AlexaPi and is started with AlexaPi.service for both Assistant and Alexa

ShaneC80, 2017-05-30:
I was out of town for a couple weeks and not been able to attempt much of anything.

In fact, I don't remember where I left off.  With that said, I'm going to try a fresh start using the AIY kit hardware but starting over with a normal Raspian Lite.

nickschwab, 2017-06-06:
I'm also encountering the same issue as @leepark (AIY Voice HAT speaker and microphone not listed as audio devices) using a clean install of Raspbian Lite and following the AssistantPi instructions.

xtools-at, 2017-06-11:
@nickschwab sorry I've been out of town for some weeks - did you manage to solve the problem? I can't really contribute here as I'm lacking the hardware :(

nickschwab, 2017-06-12:
@xtools-at No worries. I still have the problem and I'm not sure how to resolve it :-\

flemos, 2017-07-17:
Is there a formal way to use Voice Hat (that came with Google AYI kit) together with AssistantPi ?

xtools-at, 2017-07-18:
No, I'm sorry :(
I'd recommend trying to set up the AIY Kit as instructed, install [AlexaPi](https://github.com/alexa-pi/AlexaPi) (attention, _not_ AssistantPi) on top of it and see how that goes.

xtools-at, 2017-09-07:
Just found the official [instructions on how to install Voice HAT without using the image provided by Google](https://github.com/google/aiyprojects-raspbian/blob/master/HACKING.md), maybe it's helpful!

shivasiddharth, 2017-10-27:
AIY Hat has two channel input ie two microphones. The AlexaPi will work only in a single channel environment. @leepark , if two different processes need to share the audio resource then you need to use dsnoop plugin for mic else, only one service will work other will not. Try using this for asound.conf and asoundrc.. https://github.com/shivasiddharth/GassistPi/blob/master/audio-drivers/USB-DAC/scripts/asound.conf

closed 2018-04-16 by xtools-at