xtools-at/AssistantPi #10: Alexa in Failed State

goodpunk6, 2017-05-13:
Thank you for reporting an issue with **AssistantPi**.

Please make sure you've updated to the latest version using
```
sudo /opt/AlexaPi/src/scripts/update.sh
```
This updates both AssistantPi and the [tweaked Assistant SDK](https://github.com/xtools-at/assistant-sdk-python) without having you to go through the installation process again.

Please check out these resources before submitting an issue:
- [AlexaPi Wiki](https://github.com/alexa-pi/AlexaPi/wiki/)
- [AlexaPi Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/)
- [AlexaPi Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging)
- [Google Assistant Troubleshooting](https://developers.google.com/assistant/sdk/prototype/getting-started-pi-python/troubleshooting)
- [Google Assistant Audio Configuration](https://developers.google.com/assistant/sdk/prototype/getting-started-pi-python/configure-audio)

In order to help troubleshooting, be sure to include the following information:

**Python 2 & 3 releases (`python2 --version` and `python3 --version`):**
```
Whatever Jessie Lite comes with
```

**Description of problem:**
```
Alexa service is in failed state -

pi@raspberrypi:~ $ sudo systemctl status AlexaPi.service
● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
   Active: failed (Result: exit-code) since Sat 2017-05-13 21:48:10 UTC; 1min 2s ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
  Process: 1017 ExecStart=/usr/bin/python /opt/AlexaPi/src/main.py --daemon (code=exited, status=1/FAILURE)
 Main PID: 1017 (code=exited, status=1/FAILURE)

May 13 21:48:10 raspberrypi python[1017]: surround50:CARD=Device,DEV=0
May 13 21:48:10 raspberrypi python[1017]: surround51:CARD=Device,DEV=0
May 13 21:48:10 raspberrypi python[1017]: surround71:CARD=Device,DEV=0
May 13 21:48:10 raspberrypi python[1017]: iec958:CARD=Device,DEV=0
May 13 21:48:10 raspberrypi python[1017]: dmix:CARD=Device,DEV=0
May 13 21:48:10 raspberrypi python[1017]: dsnoop:CARD=Device,DEV=0
May 13 21:48:10 raspberrypi python[1017]: hw:CARD=Device,DEV=0
May 13 21:48:10 raspberrypi python[1017]: plughw:CARD=Device,DEV=0
May 13 21:48:10 raspberrypi systemd[1]: AlexaPi.service: main process exited, code=exited, status=1/FAILURE
May 13 21:48:10 raspberrypi systemd[1]: Unit AlexaPi.service entered failed state.

```

**Expected:**
```
to not be in a failed state.
```

**Steps to reproduce:**

1. follow instructions to install assistant pi
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```
reformatted SD card. copied latest stock image of jessie lite to SD card. installed assistant pi. 
```

xtools-at, 2017-05-14:
your sound config seems not be right, the setup only applies the default config. Please see the links in the README and adjust your config.yaml according to your audio in- and output devices (`aplay -L && arecord -L`).
Try running AssistantPi in debug-mode (`python /opt/AlexaPi/src/main.py -d`) for the full output, read what the error message says and act accordingly.

goodpunk6, 2017-05-14:
what was the command to remove and reinstall google manually?

xtools-at, 2017-05-14:
You can do `sudo bash /opt/AlexaPi/src/scripts/install_assistant.sh` or check out the [contents of that file](https://github.com/xtools-at/AssistantPi/blob/master/src/scripts/install_assistant.sh) to see every single command.

Basically, this should remove and re-install the SDK:
```
cd /opt/AlexaPi/src
sudo rm -rf assistant-sdk-python
sudo git clone https://github.com/xtools-at/assistant-sdk-python.git
cd /opt/AlexaPi/src/assistant-sdk-python
/opt/AlexaPi/env/bin/python -m pip install --upgrade -e ".[samples]"
```

But what are you trying to accomplish? This problem is related to the sound config of AlexaPi, not to Google Assistant SDK.

goodpunk6, 2017-05-14:
wow! thanks for your script! I went ahead and avoided all of these
headaches by installing jessie with pixel desktop. google authorization was
a breeze as the auth url was always localhost:8080. doing it on the
raspberry pi was so easy I feel terrible for bothering you.

Thanks again for this awesome script!

closed 2017-05-16 by goodpunk6