xtools-at/AssistantPi #6: Connection Refused

goodpunk6, 2017-05-13:
Thank you for reporting an issue with **AssistantPi**.

Please make sure you've updated to the latest version using
```
sudo /opt/AlexaPi/src/scripts/update.sh
```
This updates both AssistantPi and the [tweaked Assistant SDK](https://github.com/xtools-at/assistant-sdk-python) without having you to go through the installation process again.

Please check out these resources before submitting an issue:
- [AlexaPi Wiki](https://github.com/alexa-pi/AlexaPi/wiki/)
- [AlexaPi Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/)
- [AlexaPi Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging)
- [Google Assistant Troubleshooting](https://developers.google.com/assistant/sdk/prototype/getting-started-pi-python/troubleshooting)
- [Google Assistant Audio Configuration](https://developers.google.com/assistant/sdk/prototype/getting-started-pi-python/configure-audio)

In order to help troubleshooting, be sure to include the following information:

**Python 2 & 3 releases (`python2 --version` and `python3 --version`):**
```
Whatever comes with Raspbian Lite
```

**Description of problem:**
```
When i try to authorize Google stuff, I'm brought to a website that asks me to Allow google's api (i think). When I click allow, nothing happens. I'm brought to a grey page that simply says connection refused. picture here - https://dl.dropboxusercontent.com/s/zfuir6sri5hcxms/chrome_2017-05-12_22-54-50.png

![chrome_2017-05-12_22-54-50 1](https://cloud.githubusercontent.com/assets/15090661/26021978/b53dff72-3766-11e7-9ea2-13cfbf54b6f2.png)

```

**Expected:**
```
to authorize.
```

**Steps to reproduce:**

1. attempt to authenticate with google
2. click allow
3. look at connection refused page.

**Traceback (if applicable):**
```

```

**Additional info:**
```
I have a raspberry pi 3 that has nothing else but Assistant Pi on it.
```

xtools-at, 2017-05-13:
hi, please provide more information - is it sure the google auth or is it the amazon authorization? please paste the lines that came before this in your terminal and provide the url you are trying to open.

Sidenote: I'm re-writing AssistantPi right now, as I was experiencing some painful bugs. Better wait a few days for the new release.

goodpunk6, 2017-05-13:
I had to start the authorization manually. 

```
pi@raspberrypi:~ $ /opt/AlexaPi/env/bin/python -m googlesamples.assistant.auth_helpers --client-secrets /home/pi/Downloads/client_secret.json
Please visit this URL to authorize this application: https://accounts.google.com/o/oauth2/auth?response_type=code&client_id=1071494548268-9bf77r8k5kkd3hts6jpij5h62hdreqh3.apps.googleusercontent.com&redirect_uri=http%3A%2F%2Flocalhost%3A8080%2F&scope=https%3A%2F%2Fwww.googleapis.com%2Fauth%2Fassistant-sdk-prototype&state=cnfplcb0wjZqT7CCo9bN9QOzzSDr17&access_type=offline

```

xtools-at, 2017-05-13:
I can open the link without problems, no "Connection refused" - maybe you could try to disable your Adblocker or try another Browser.

goodpunk6, 2017-05-13:
I did disable my firewall. I'll try another browser

goodpunk6, 2017-05-13:
i disabled my firewall and tried 3 different browsers. I still get connection refused.

goodpunk6, 2017-05-14:
I formatted and reinstalled at least 6 times. everytime, connection refused.

xtools-at, 2017-05-14:
It has nothing to do with AssistantPi that you can't open Google's Auth link, there's something about your setup/network... plz try open the link on another computer, on your phone or any other device that has a Browser - I can't help you with this

closed 2017-05-14 by xtools-at

goodpunk6, 2017-05-14:
i have tried on 3 different networks. still happens. wtf? please reopen issue. not my networks.

xtools-at, 2017-05-14:
Seriously, I can't help here. The [link](https://accounts.google.com/o/oauth2/auth?response_type=code&client_id=1071494548268-9bf77r8k5kkd3hts6jpij5h62hdreqh3.apps.googleusercontent.com&redirect_uri=http%3A%2F%2Flocalhost%3A8080%2F&scope=https%3A%2F%2Fwww.googleapis.com%2Fauth%2Fassistant-sdk-prototype&state=cnfplcb0wjZqT7CCo9bN9QOzzSDr17&access_type=offline) you've posted works fine:
[image]

Additionally, I haven't altered the Authorization with Google API, it's 100% the same as in the SDK. If you can't get it to work, please file an issue in [Google's Assistant SDK Repository](https://github.com/googlesamples/assistant-sdk-python).

goodpunk6, 2017-05-14:
Ok thanks!