xtools-at/AssistantPi #8: AlexaPi doesn't work over wifi

goodpunk6, 2017-05-13:
Thank you for reporting an issue with **AssistantPi**.

Please make sure you've updated to the latest version using
```
sudo /opt/AlexaPi/src/scripts/update.sh
```
This updates both AssistantPi and the [tweaked Assistant SDK](https://github.com/xtools-at/assistant-sdk-python) without having you to go through the installation process again.

Please check out these resources before submitting an issue:
- [AlexaPi Wiki](https://github.com/alexa-pi/AlexaPi/wiki/)
- [AlexaPi Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/)
- [AlexaPi Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging)
- [Google Assistant Troubleshooting](https://developers.google.com/assistant/sdk/prototype/getting-started-pi-python/troubleshooting)
- [Google Assistant Audio Configuration](https://developers.google.com/assistant/sdk/prototype/getting-started-pi-python/configure-audio)

In order to help troubleshooting, be sure to include the following information:

**Python 2 & 3 releases (`python2 --version` and `python3 --version`):**
```
whatever comes default with Jessie Lite
```

**Description of problem:**
```
AlexaPi doesn't work over wifi
```

**Expected:**
```
AlexaPi should work over wifi
```

**Steps to reproduce:**

1. install assistant pi while connected via eth0 (wired)
2. complete setup and reboot
3. check status pi@raspberrypi:~ $ sudo systemctl status AlexaPi.service
Hint: Some lines were ellipsized, use -l to show in full.

**Traceback (if applicable):**
```

```

**Additional info:**
```
pi@raspberrypi:~ $ sudo systemctl status AlexaPi.service
● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
   Active: active (running) since Sat 2017-05-13 17:29:53 UTC; 5min ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
 Main PID: 1420 (python)
   CGroup: /system.slice/AlexaPi.service
           └─1420 /usr/bin/python /opt/AlexaPi/src/main.py --daemon

May 13 17:33:27 raspberrypi python[1420]: self._send_output(message_body)
May 13 17:33:27 raspberrypi python[1420]: File "/usr/lib/python2.7/httplib.py", line 879, in _send_output
May 13 17:33:27 raspberrypi python[1420]: self.send(msg)
May 13 17:33:27 raspberrypi python[1420]: File "/usr/lib/python2.7/httplib.py", line 841, in send
May 13 17:33:27 raspberrypi python[1420]: self.connect()
May 13 17:33:27 raspberrypi python[1420]: File "/usr/local/lib/python2.7/dist-packages/requests/packages/urllib3/connect...onnect
May 13 17:33:27 raspberrypi python[1420]: conn = self._new_conn()
May 13 17:33:27 raspberrypi python[1420]: File "/usr/local/lib/python2.7/dist-packages/requests/packages/urllib3/connect...w_conn
May 13 17:33:27 raspberrypi python[1420]: self, "Failed to establish a new connection: %s" % e)
May 13 17:33:27 raspberrypi python[1420]: NewConnectionError: <requests.packages.urllib3.connection.VerifiedHTTPSConnect... known
Hint: Some lines were ellipsized, use -l to show in full.

```

xtools-at, 2017-05-14:
I run AssistantPi solely over Wifi and hadn't had any issues yet. This looks like AlexaPi can't handle network switches during runtime, but if you boot your Pi up with either LAN or Wifi connected, it should work fine.

goodpunk6, 2017-05-14:
That's right. I have reinstalled and it works. This thread can be closed.

closed 2017-05-14 by xtools-at