xtools-at/AssistantPi #11: Alexa stops listening

goodpunk6, 2017-05-14:
Thank you for reporting an issue with **AssistantPi**.

Please make sure you've updated to the latest version using
```
sudo /opt/AlexaPi/src/scripts/update.sh
```
This updates both AssistantPi and the [tweaked Assistant SDK](https://github.com/xtools-at/assistant-sdk-python) without having you to go through the installation process again.

Please check out these resources before submitting an issue:
- [AlexaPi Wiki](https://github.com/alexa-pi/AlexaPi/wiki/)
- [AlexaPi Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/)
- [AlexaPi Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging)
- [Google Assistant Troubleshooting](https://developers.google.com/assistant/sdk/prototype/getting-started-pi-python/troubleshooting)
- [Google Assistant Audio Configuration](https://developers.google.com/assistant/sdk/prototype/getting-started-pi-python/configure-audio)

In order to help troubleshooting, be sure to include the following information:

**Python 2 & 3 releases (`python2 --version` and `python3 --version`):**
```
Whatever comes with Jessie Lite
```

**Description of problem:**
```
Alexa will stop listening after saying hello on boot.
```

**Expected:**
```
I expect Alexa to respond when I say Alexa
```

**Steps to reproduce:**

1. Install Assistant Pi with Google and airplay support
2. setup audio to pulse system wide as recommended
3. after a reboot, and after alexa says hello, call alexa

**Traceback (if applicable):**
```

```

**Additional info:**
```
python /opt/AlexaPi/src/main.py -d
INFO: pocketsphinx.c(152): Parsed model-specific feature parameters from /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/feat.params
Current configuration:
[NAME]                  [DEFLT]         [VALUE]
-agc                    none            none
-agcthresh              2.0             2.000000e+00
-allphone
-allphone_ci            no              no
-alpha                  0.97            9.700000e-01
-ascale                 20.0            2.000000e+01
-aw                     1               1
-backtrace              no              no
-beam                   1e-48           1.000000e-48
-bestpath               yes             yes
-bestpathlw             9.5             9.500000e+00
-ceplen                 13              13
-cmn                    live            batch
-cmninit                40,3,-1         41.00,-5.29,-0.12,5.09,2.48,-4.07,-1.37,-1.78,-5.08,-2.05,-6.45,-1.42,1.17
-compallsen             no              no
-debug                                  0
-dict                                   /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/cmudict-en-us.dict
-dictcase               no              no
-dither                 no              no
-doublebw               no              no
-ds                     1               1
[... 173 lines cut ...]
buffer (last 100 chars): ''
before (last 100 chars): 'opt/AlexaPi/src/scripts/auth_assistant.sh  to initialize new OAuth2 credentials.\r\nassistant_record\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: 0
flag_eof: True
pid: 957
child_fd: 13
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: None
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile("Recording audio .*")

```

xtools-at, 2017-05-14:
The logs are telling us that the script has crashed - it isn't listening anymore as the app isn't running.

I can even see *why* it crashed:

It starts with
```
2017-05-14 14:51:02 DEBUG: Pocketsphinx Trigger found voice command: **google**
2017-05-14 14:51:02 INFO: Starting Assistant conversation
```
The hotword recognition (mis-?)heard your command and triggered Assistant. Then it fails, and in the Traceback, it says it got stuck on Assistant complaining about
```
... initialize new OAuth2 credentials
```
You haven't done the Authorization with the Google API, this is why the script crashes when Assistant is triggered.

xtools-at, 2017-05-15:
Hi there,
just a quick hint: I've completed the 1.0 release today, an important fix is included. Please run the Updater script or Assistant might crash otherwise.

closed 2017-05-16 by goodpunk6