xtools-at/AssistantPi #12: it crash after running for awhile

ttruong15, 2017-05-16:
Thank you for reporting an issue with **AssistantPi**.

Please make sure you've updated to the latest version using
```
sudo /opt/AlexaPi/src/scripts/update.sh
```
This updates both AssistantPi and the [tweaked Assistant SDK](https://github.com/xtools-at/assistant-sdk-python) without having you to go through the installation process again.

Please check out these resources before submitting an issue:
- [AlexaPi Wiki](https://github.com/alexa-pi/AlexaPi/wiki/)
- [AlexaPi Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/)
- [AlexaPi Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging)
- [Google Assistant Troubleshooting](https://developers.google.com/assistant/sdk/prototype/getting-started-pi-python/troubleshooting)
- [Google Assistant Audio Configuration](https://developers.google.com/assistant/sdk/prototype/getting-started-pi-python/configure-audio)

In order to help troubleshooting, be sure to include the following information:

**Python 2 & 3 releases (`python2 --version` and `python3 --version`):**
```
Python 2.7.9
```

**Description of problem:**
```
Seem to crash after the system run for awhile.  Could be 10mins or an hour but always crash with the following log status:

sudo systemctl status Alexa.service:

May 17 07:51:06 raspberrypi3 python[4843]: INFO: Assistant conversation finished
May 17 07:51:08 raspberrypi3 python[4843]: Exception in thread Thread-1:
May 17 07:51:08 raspberrypi3 python[4843]: Traceback (most recent call last):
May 17 07:51:08 raspberrypi3 python[4843]: File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
May 17 07:51:08 raspberrypi3 python[4843]: self.run()
May 17 07:51:08 raspberrypi3 python[4843]: File "/usr/lib/python2.7/threading.py", line 763, in run
May 17 07:51:08 raspberrypi3 python[4843]: self.__target(*self.__args, **self.__kwargs)
May 17 07:51:08 raspberrypi3 python[4843]: File "/opt/AlexaPi/src/alexapi/triggers/pocketsphinxtrigger.py", line 95, in thread
May 17 07:51:08 raspberrypi3 python[4843]: voice_command = self._decoder.hyp().hypstr
May 17 07:51:08 raspberrypi3 python[4843]: AttributeError: 'NoneType' object has no attribute 'hypstr'

```

**Expected:**
```

```

**Steps to reproduce:**

1.
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

xtools-at, 2017-05-17:
Thanks for reporting, it seems that pocketsphinx here and there returns a null-Object when recognizing hotwords. I'll fix this soon!

xtools-at, 2017-05-17:
just pushed release 1.1, the issue is fixed there.
please run the updater and try again!

xtools-at, 2017-05-20:
please reopen if the problem still occurs!

closed 2017-05-20 by xtools-at

xtools-at, 2017-06-11:
just for the record, I've set up an instance on a Raspi Zero W and it's running fine for 24+ hours in my bathroom :)