xtools-at/AssistantPi #13: AlexaPi is loud & clear but Google Assistant is too soft.

rsuhendro, 2017-05-24:
Thank you for reporting an issue with **AssistantPi**.

Please make sure you've updated to the latest version using
```
sudo /opt/AlexaPi/src/scripts/update.sh
```
This updates both AssistantPi and the [tweaked Assistant SDK](https://github.com/xtools-at/assistant-sdk-python) without having you to go through the installation process again.

Please check out these resources before submitting an issue:
- [AlexaPi Wiki](https://github.com/alexa-pi/AlexaPi/wiki/)
- [AlexaPi Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/)
- [AlexaPi Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging)
- [Google Assistant Troubleshooting](https://developers.google.com/assistant/sdk/prototype/getting-started-pi-python/troubleshooting)
- [Google Assistant Audio Configuration](https://developers.google.com/assistant/sdk/prototype/getting-started-pi-python/configure-audio)

In order to help troubleshooting, be sure to include the following information:

**Python 2 & 3 releases (`python2 --version` and `python3 --version`):**
```
**Python 2.7.9 & Python 3.4.2**

```

**Description of problem:**
```
The voice output level of Google Assistant is too low. Sometimes the hotword **Google** was also missed, no triggering.

```

**Expected:**
```
The same level as the AlexaPi.

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```
Run on RPI 3 with Raspbian Jessie Lite. I use PulseAudio in the config.yaml because for unknown reason I can't make AlexaPi works with ALSA.
I've set the volume level using alsamixer. 
```

xtools-at, 2017-05-24:
Hi,

have you tried setting the output volume by using Google's and Alexa's native voice commands? For Alexa, this works (it starts with a default of ~80%). [The Google SDK claims to suport voice control](https://github.com/googlesamples/assistant-sdk-python/tree/master/google-assistant-sdk/googlesamples/assistant/grpc#readme) for volume, but I haven't tested it yet.

Just fyi, AssistantPi defaults to using pulseaudio since release 1.0 and does all necessary settings during the setup.

rsuhendro, 2017-05-25:
I've tried to reduce default_volume to 40% in config.yaml but Alexa is still loud. I can't find the volume control command for Google SDK. 
Any hints?

AISense-UTM, 2017-06-02:
Same issue with a RPI 2, has anyone been able to increase the volume for Google SDK. It is way too low if compare with Alexa.

bote795, 2017-06-03:
found the voice command to increase googles voice, https://support.google.com/googlehome/answer/7072489?hl=en

rsuhendro, 2017-06-10:
Thanks. It works. I can set max volume (level 10). But somehow Alexa is still louder, so I need to find a way to lower it.

xtools-at, 2017-06-11:
Sorry, I've been out of town for some days - you can adjust Alexa's volume with it's native voice command, simply say "Alexa, louder/softer" or "Alexa, Voume 4"
[https://www.amazon.com/gp/help/customer/display.html?nodeId=201619520](https://www.amazon.com/gp/help/customer/display.html?nodeId=201619520)

rsuhendro, 2017-06-12:
Just perfect...fully solved. Thanks a lot.

xtools-at, 2017-06-14:
It seems that adjusting Alexa's volume isn't working properly (RPi Zero W). I get the message in debug mode, that volume has been adjusted, but there's no difference between 2 and 10... Anyone else experiencing this issue?

zerocolTHS, 2017-06-17:
I am also facing the same issue, no matter what i try alexa i always so much louder. The volume doesnt seem to change.

I also have an issue with google activating at almost any word, even if its the tv it activates and starts talking. 

Great work though.

rsuhendro, 2017-06-18:
Yes the command works: Volume 4
2017-06-18 22:38:51 DEBUG: JSON String Returned: {
  "messageBody": {
    "directives": [
      {
        "namespace": "Speaker", 
        "name": "SetVolume", 
        "payload": {
          "volume": 40, 
          "adjustmentType": "absolute"
        }
      }
    ]
  }, 
  "messageHeader": {}
}
2017-06-18 22:38:51 DEBUG: Setting volume to: 40
2017-06-18 22:38:51 DEBUG: new volume = 40

But the voice level is as loud as before..

xtools-at, 2017-07-14:
tried to hack around this problem by setting the system volume via subprocess/pexpect using `amixer -D pulse sset Master 50%`, but it seems to have no effect on the daemon, even if it runs in the same scope (i.e. run by service, not manually by a Linux user account). Anyone any ideas on this?