xtools-at/AssistantPi #14: urllib3 problem

mcosti, 2017-06-02:
Hello. When I try to activate my Google Credentials, this is what I get.
   raise RuntimeError('Requests dependency \'urllib3\' must be version >= 1.21.1, < 1.22!')
RuntimeError: Requests dependency 'urllib3' must be version >= 1.21.1, < 1.22!

I tried to upgrade it to 1.21.1, but failed.

Thank you for your help.

bote795, 2017-06-03:
same issue... though part of error also says following : 
`"/usr/lib/python3.4/importlib/util.py"`

but my python is 2.7.9 not sure if pip installs are version dependent

bote795, 2017-06-03:
try runing the following @mcosti 
` sudo /opt/AlexaPi/env/bin/python -m pip install -I urllib3==1.21.1`

fixed it for me.. went into the install script and saw that it created an env so have to install into that env

mcosti, 2017-06-03:
This worked. Thank you.

closed 2017-06-03 by mcosti

mcosti, 2017-06-03:
Oh. Actually, it worked on the Authentication. But on this:
sudo bash /opt/AlexaPi/src/scripts/install_assistant.sh

Same error.

bote795, 2017-06-03:
yea that reruns all the installation packages so you need to rerun the command I said, and you only need to run the authentication one (it comes out when you run the command you said). instead of the whole installation

xtools-at, 2017-06-11:
Sorry, I've been out of town for some days - Glad you found the solution, I'll include this into the installer script!