xtools-at/AssistantPi #15: Unable to authenticate Alexa due to lack of web browser.

Scardaddy, 2017-06-17:
Thank you for reporting an issue with **AssistantPi**.

Please make sure you've updated to the latest version using
```
sudo /opt/AlexaPi/src/scripts/update.sh
```
This updates both AssistantPi and the [tweaked Assistant SDK](https://github.com/xtools-at/assistant-sdk-python) without having you to go through the installation process again.

Please check out these resources before submitting an issue:
- [AlexaPi Wiki](https://github.com/alexa-pi/AlexaPi/wiki/)
- [AlexaPi Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/)
- [AlexaPi Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging)
- [Google Assistant Troubleshooting](https://developers.google.com/assistant/sdk/prototype/getting-started-pi-python/troubleshooting)
- [Google Assistant Audio Configuration](https://developers.google.com/assistant/sdk/prototype/getting-started-pi-python/configure-audio)

In order to help troubleshooting, be sure to include the following information:

**Python 2 & 3 releases (`python2 --version` and `python3 --version`):**
```
2.7.9
```

**Description of problem:**
```
unable to authenticate Alexa service without web browser on jessie lite
```

**Expected:**
```
Setup to finish with authentication
```

**Steps to reproduce:**

1. vanilla jessie lite install
2. follow steps in your tutorial
3. try to authenticate Alexa VS

**Traceback (if applicable):**
```
Amazons page says 
```
After installing a browser this is the message that is provided by amazon: 

We're sorry!
An error occurred when we tried to process your request. Rest assured, we're already working on the problem and expect to resolve it shortly. 

**Additional info:**
```

```

xtools-at, 2017-06-18:
First, you've to figure out what IP your RasPi got _within_ **your** network (e.g. 192.167.0.105; check with `iwconfig` or `ipconfig` or in your Router's Dashboard).
Then insert that IP in your developer.amazon.com-Settings as **Allowed Origin** as described [here](https://github.com/alexa-pi/AlexaPi/wiki/Installation) in Step 1, last bulet point.
Afterwards, you should be able to do the Authentication on any device (e.g. your Laptop), if it's connected to the _same_ network.

Scardaddy, 2017-06-18:
That does not work.  I have tried it in the way I described, and the way you described.  The result is exactly the same.  I will happily make a video for you to prove it to be the case.  I will post a link latter on today to confirm.  I do apologize if it is simply user error, but I am pretty sure having run through this multiple times now that it is simply not the case.  The internal IP addresses are set correctly and the information is correctly inserted into the AVS developer settings.

Scardaddy, 2017-06-19:
Having just run through a total clean attempt at a reinstall from scratch I ran into two issues this being one of them, so it is still a problem.  One questions, how to copy and paste the amazon credentials to the command line so as not to enter incorrect information.  Aside from that git is not installed by default on Jessie lite so there's an immediate failure on that step.  I'm not trying to break it, I'd actually really like it to work.  But it's broken, either because I am missing something not mentioned or because it can't work as written.  Thank you for your help and hard work so far Martin, and hopefully it is just user error.

xtools-at, 2017-06-19:
Sorry I didn't meant to be rude - I tag all issues for myself to be able to broadly categorize them. And I really thought my advice would help you solve your issue, sadly it didn't - it could have been so easy :(

I don't have any ideas right now how to solve the problem, give me some time and I'll maybe figure something out!

Is the latest Jessie Lite really lacking Git? Never had an issue with that, have to run a fresh install once more to check.

Scardaddy, 2017-06-19:
No need to apologies you have put in so much work already, and I would be a jerk to be anything less then appreciative.  I'll keep on trying to find a solution or at the very least narrow down the problem. I'll update when and if I get somewhere.  Perhaps there is something in the chosen naming convention for the AVS credentials.  But not sure, I'll keep on trying.  Thank you again for your help and hard work. :)

xtools-at, 2017-07-14:
any news on this one?

Scardaddy, 2017-07-14:
I've got nothing so far.   But to be fair I haven't had much chance to look
at it. We're you able to perform a fresh install?

Jonher937, 2017-07-29:
Post the output from:
`ip a; ip l`
As well as a screenshot from the Amazon developer portal -> Alexa -> Edit Device -> Security Profile -> Web Settings (take screenshot of the Allowed origins as well as Return URLs). 
The URL you get from the script needs to be an exact match of the Allowed Origins.

xtools-at, 2017-08-20:
I'll close this in the meantime to unclutter the issues, please re-open if you still need help!

closed 2017-08-20 by xtools-at

Scardaddy, 2017-08-20:
I'd be happy to submit a more detailed explanation.  ip a and ip l as was
requested offer no insight into what the problem is.  Attached are again
the results of following your instructions as layed out on the projects
home page.  Jessie is no longer the current build for Raspbian, I have not
tested the new lite version of stretch.  I am not sure if you can try again
to follow your own steps on a fresh install.  Again I still have the same
errors, I can not authenticate the AVS service on any device on my
network.  Enclosed are screen shots of the output of ip a and ip l from the
Pi, they both show the same address as setup in the AVS service pages.  I
assumed that the problem was associated to entering long strings of letters
and numbers but after triple checking that is not the case.  My last idea
is that it is a network issue and that there needs to be some ports
forwarded to enable this to work.  But that is the type of information I
would expect you to have already run into if it was the case.  I am
guessing by closing the ticket you have neither found the problem I am
having, nor have tried to reinstall from scratch.  I definitely don't know
enough to tell you specifically what the problem is, so if I don't hear
back I will assume there is no resolution.

Thanks for your help and work on the project.
[image: Inline images 4][image: Inline images 2][image: Inline images 3]

Scardaddy, 2017-08-20:
I'd like to point out that I have followed other instructions to get Alexa to authenticate on Raspbian, however it was with PIXEL and therefore failed your requirements.  So this problem is very much specific to this project.

Jonher937, 2017-08-20:
@Scardaddy Looks like we can't see the screenshots attached because you replied by email. I tried a fresh install 3days ago and that has been working fine on the latest Raspbian Stretch Lite image.
Please post screenshots directly on the github issue page and we'll have a look.

Scardaddy, 2017-08-20:
[image]
[image]
[image]

Scardaddy, 2017-08-20:
[image]

Jonher937, 2017-08-20:
The IP's indeed looks OK. 
I guess you have double-checked the ID's/keys that are entered info the AlexaPi config/setup? 

Attaching screenshots from my working setup, make sure you are entering/copying the exact strings and remember that all of the fields are case sensitive.
I got the same type of errors when I had the wrong "Device_Type_ID" or "Security_Profile_ID".

If that does not do it I would recreate the Alexa app in the developers settings and rerun the AlexaPI setup with the new ID's. 
[image]
[image]

Scardaddy, 2017-08-20:
Thank you, I will try again from scratch focusing on the points you
mentioned.  I'll reply, hopefully with positive results.

Scardaddy, 2017-08-22:
I tried again with the same results.  It seems the naming convention has changed between what the script asks for and what Amazon now calls things.  I got a little confused and perhaps that with the requirement for names to not be the same but no specification on which can not be the same.  In the end I tried every name different.  However, in the screen shot you added, not every thing that is necessary is filled in.  I have attached my settings with the obvious redacted information, perhaps something obvious stands out to you.  At this point I can only assume its an incorrect setting either in the amazon settings or an incorrect redirect due to a complicated internal network.  I also attached a more detailed error page from Amazon, I doubt that will help aside from the obvious.
[image]
[image]
[image]

Jonher937, 2017-08-22:
Hello, please edit the return urls to include "/code" at the end.
```
http://192.168.1.35:5050/code
http://localhost:5050/code
```

Scardaddy, 2017-08-22:
Oh wow, I can't believe after all that, it was just /code.  Authentication works no problem.  Was the /code implemented recently.  I fail to remember seeing it before you mentioned it, but looking back it was in there.  Thank you for your help Jonher, I wouldn't say its working perfectly, but I am now one step further in the list.  Thanks again.