xtools-at/AssistantPi #16: Setup failure for pip in script

hotdogh2o, 2017-06-22:
**Description of problem:**
```Searching for pip
Reading https://pypi.python.org/simple/pip/
Download error on https://pypi.python.org/simple/pip/: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:581) -- Some packages may not be found!
Couldn't find index page for 'pip' (maybe misspelled?)
Scanning index of all packages (this may take a while)
Reading https://pypi.python.org/simple/
Download error on https://pypi.python.org/simple/: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:581) -- Some packages may not be found!
No local packages or download links found for pip
error: Could not find suitable distribution for Requirement.parse('pip')

```

**Expected:**
```

```

**Steps to reproduce:**

1. install fresh jessie lite
2. sudo apt-get update
3. sudo apt-get dist-upgrade
4.sudo apt-get install git
5.cd /opt
6.sudo git clone https://github.com/xtools-at/AssistantPi.git AlexaPi
7.sudo /opt/AlexaPi/src/scripts/setup.sh

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

xtools-at, 2017-07-13:
Hi there,
sorry I've been busy, got a new job a few weeks ago. Is this still an issue? From the problem description, I assume that https://pypi.python.org had SSL issues that day :(

lvidarte, 2017-07-24:
Same problem here

lvidarte, 2017-07-24:
Date Issue (too old):

    $ date
    Mon Apr 10 11:30:04 UTC 2017

Update:

    $ sudo date -s "24 JUL 2017 12:00:00"
    Mon Jul 24 12:00:00 UTC 2017

Now it works:

    $ sudo python2 -m easy_install pip

xtools-at, 2017-07-25:
thanks @lvidarte for verifying! any idea how I could fix this?

lvidarte, 2017-07-25:
From ntp man page: "ntp doesn't update the time if the offset exceeds the panic threshold, which is 1000 s by default."

This is the best way I found to update time

    sudo systemctl stop ntp
    sudo ntpd -gq
    sudo systemctl start ntp

The -gq tells the ntp daemon to correct the time regardless of the offset (g) and exit immediately (q) after setting the time.

hotdogh2o, 2017-07-25:
It was the time! thanks everyone for your help!

closed 2017-07-25 by hotdogh2o

xtools-at, 2017-07-26:
@lvidarte thanks for pointing out, I thought it was because the certificate expired and didn't get that it was the time on the device itself.
I'll include this in the next Release, thanks again!

xtools-at, 2017-07-28:
Version 1.2 released, shouldn't be an issue to anyone anymore.