xtools-at/AssistantPi #17: Service crash, right after first and slow Google Assistant query

0rionas, 2017-07-05:
The service starts and "Hello" is being heard. Alexa works fine and answers everything. When "Google" is said, it takes from 30 seconds to 1 minute to hear the beep, the answer to the question comes up, then the service hangs with the following error.

```
Jul 05 07:25:02 raspberrypi python[1769]: INFO: Assistant conversation finished
Jul 05 07:25:02 raspberrypi python[1769]: Exception in thread Thread-1:
Jul 05 07:25:02 raspberrypi python[1769]: Traceback (most recent call last):
Jul 05 07:25:02 raspberrypi python[1769]: File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
Jul 05 07:25:02 raspberrypi python[1769]: self.run()
Jul 05 07:25:02 raspberrypi python[1769]: File "/usr/lib/python2.7/threading.py", line 763, in run
Jul 05 07:25:02 raspberrypi python[1769]: self.__target(*self.__args, **self.__kwargs)
Jul 05 07:25:02 raspberrypi python[1769]: File "/opt/AlexaPi/src/alexapi/triggers/pocketsphinxtrigger.py", line 61, in thread
Jul 05 07:25:02 raspberrypi python[1769]: inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, self._config['sound']['input_device'])
Jul 05 07:25:02 raspberrypi python[1769]: ALSAAudioError: Device or resource busy [plughw:CARD=Set,DEV=0]
```

I have been doing tests with both PS3 Eye and a USB Sound card. The results are pretty much the same. Alexa seems to be working no problem in all cases. Google Assistant works fine with the "assistant_record" command of the SDK. But service hangs once Google Assistant is invoked once.

config.yaml has its input set to plughw. Am I supposed to pick something else?

Thanks in advance

0rionas, 2017-07-11:
@xtools-at I sincerely hope this project is not dead. I haven't found anything like it around, and I just love the concept. I have my fingers crossed that you are on vacation and that you will respond to these messages in a few days.

xtools-at, 2017-07-13:
@0rionas sorry I've been busy, got a new job a few weeks ago and doing coding 8h+ a day on other projects.

what got me suspicious is the last line in your error log:
`raspberrypi python[1769]: ALSAAudioError: Device or resource busy [plughw:CARD=Set,DEV=0]`

this means the "beep" and Assistant's output are conflicting, i.e. they're trying to play simultanously - which shouldn't happen, as the beep should be long done when Assistant's done recording and starts playing the response...

I tried to mitigate the problem. Could you please try the following?
```
cd /opt/AlexaPi
sudo git checkout .
sudo git checkout feature/google_timing_issues
```
then restart the service or script or reboot the pi and let me know if this fixes the issue!

xtools-at, 2017-07-25:
it's really weird though - I have AssistantPi running on a Pi Zero, which is really really slow, and hadn't had this problem before :(

nmatthews1, 2017-08-15:
Same exact issue here. I am able to trigger the assistant with the wake word, and there is no beep at all, but i am able to ask and receive a response, immediately followed by this error.

The git checkout just gave an error 

error: pathspec 'feature/google_timing_issues' did not match any file(s) known to git

I hope there's a solution to this. If it really is just the beep causing the issue, can it be disabled?

2017-08-15 05:05:21 INFO: Checking Internet Connection ...
2017-08-15 05:05:21 INFO: Connection OK
2017-08-15 05:05:21 INFO: AVS token: Requesting a new one
2017-08-15 05:05:22 INFO: AVS token: Obtained successfully
2017-08-15 05:05:38 INFO: Starting Assistant conversation
2017-08-15 05:05:51 INFO: Assistant conversation finished
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/opt/AlexaPi/src/alexapi/triggers/pocketsphinxtrigger.py", line 61, in thread
    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, self._config['sound']['input_device'])
ALSAAudioError: Device or resource busy [default]

nmatthews1, 2017-08-15:
For further info, I can start a conversation fine and it will work back and forth as long as the conversation stays open. It is only when the conversation is finished that it errors. It is the finishing of the conversation that is triggering whatever the issue is

xtools-at, 2017-08-20:
as of the checkout of the experimental branch: it's definitely there, see [https://github.com/xtools-at/AssistantPi/tree/feature/google_timig_issue](https://github.com/xtools-at/AssistantPi/tree/feature/google_timig_issue), don't know what's causing this issue for you :(

but you could try the following to disable the beep altogether and re-test this:
- (stay on the master branch)
- stop AlexaPi service with `sudo systemctl stop AlexaPi.service`
- on your Pi, go find [opt/AlexaPi/src/main.py](https://github.com/xtools-at/AssistantPi/blob/master/src/main.py) and comment out lines 359 (`sound = pexpect.spawn('cvlc /opt/AlexaPi/src/resources/okgoogle.mp3')`) and 363 (`sound.close(force=True)`)
- restart AssistantPi with `sudo systemctl start AlexaPi.service`

please let me know if the issue still appears!

Scardaddy, 2017-08-25:
I actually have the same issue.  I have tried to comment out lines 359 and 363, they did not fix the problem.  However, it did remove the audible tone.  In debug I find that the google assistants response to the hotword to be far too short.  It is irregular but sometimes lasting less then a tenth of a second before closing.  If I say the hotword then continue speaking as quickly as possible I can keep the assistant from closing.  However, it will crash after providing an answer, forcing me to ctrlC the script.

xtools-at, 2017-09-07:
I'll try to recreate that on my Pi! If it's not the sound causing the concurrent access to the audio resources, I really don't have a clue right now :(

Scardaddy, 2017-09-07:
The only thing that comes to mind is latency in the hardware capture
process.  The two devices I have tried are:

Kinobo - USB 2.0 Mini Microphone
<https://www.amazon.com/gp/product/B00IR8R7WQ/ref=oh_aui_search_detailpage?ie=UTF8&psc=1>

and

Creative Live!
<https://www.amazon.com/gp/product/B004431UBM/ref=oh_aui_search_detailpage?ie=UTF8&psc=1>

Not sure iof it would impact or if the USB mic was pretty much the same as
everyone else uses.

I'll keep looking and thanks for your help.

MarkGrenville, 2017-10-05:
This is happening to me too. When Google responds the script crashes. 

```
Exception in thread Thread-1:
Traceback (most recent call last):
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/opt/AlexaPi/src/alexapi/triggers/pocketsphinxtrigger.py", line 61, in thread
    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, self._config['sound']['input_device'])
ALSAAudioError: Device or resource busy [mic]

```
Also It often happens to me that Google will start assistant conversation and close it instantly.

Also there is quite a delay between Google saying starting conversation the the beep. 

Will try spend some time on it this weekend and see what I can find.

xtools-at, 2017-10-14:
Just saw this change in the [Assistant SDK Release Notes
](https://developers.google.com/assistant/sdk/release-notes), might be related:
> The googlesamples-assistant-hotword sample gives a connection error. This is an issue with slow or spotty internet connections; try to connect to a faster/more reliable network.
> This will be fixed in the upcoming release.

Scardaddy, 2017-10-14:
I was using a wired connection to a solid network. Unless the three
different Pi 3s I tried had hardware problems, then this wasn't the source
of the problem for me specifically.  Though it might be for others.

Thank you for the help,

coindropper, 2017-10-20:
Same issue for me as well. Once google or Alexa respond, you can hear the speakers never shut off and I get the device or resource busy error. I can install alexapi or google assistant separately and run them without issue (Just not at the same time, obviously).

dpoulter2, 2017-12-13:
Hey @xtools-at is there any solution to this? I am getting the same error. Thanks

xtools-at, 2018-04-16:
may be related to [this](https://github.com/googlesamples/assistant-sdk-python/issues/230)