xtools-at/AssistantPi #18: Suggestion: Simple Status monitoring API

eouia, 2017-07-20:
Thanks for your great Job.

This issue is not for the bug, rather than a suggestion.

I want to use this module for my homebrew project - magic mirror. It has no good input interface except voice or some kinds of sensor. So I believe this voice assistant could be good option for me. (Of course, I shall make some Alexa Skills or Google Actions for controlling my mirror)

Anyway,
There is a little missing feature to use this module with other program.
It is very difficult to detect this program's working well or not in background process. Is it alive? It is normally working but only speaker volume is just too low to hear? When on development stage, I can check everything by myself. But after deploying, It's not easy. My main users(my family) wouldn't like if Alexa or Google doesn't answer when they call. 

 I want to monitor this service status by other program. Maybe I could catch the logs. But if you provide a small feature to broadcast signal with simple REST API, It could be more convenient.

It would be simple. 
You just send a tiny Heartbeat signal to pre-configurable(in `config.YAML`??) HTTP endpoint. I will catch the signal and can guess the status of program by it. 
By example, 
Assuming this URL for listener.
`http://mymonitor.server/listen`
And you can send some signals to this API occasionally or schedully.
when started : POST `{status:'started'}`
when alive, every 1 minutes(if available) : POST `{status:alive}`
when hotwords detected: POST `{status:actiavted}`
when Assistants works : POST `{status:processing}`
when Assistants failed : POST `{status:failed}`
and so on...
...

And I just catch this HTTP Request and use for variable things, like showing status on my mirror or displaying warning on inactive, or using as trigger for other jobs(e.g. rebooting, stop using another mic or speaker)

Could you provide this features? I'm not a expert so I cannot make PULL REQUEST by myself.

xtools-at, 2017-07-25:
Hi @eouia,
sorry but I can't do this for you.
You can try play around with AlexaPi's pre- and post-interaction hooks. In the `config.yaml`, you'll find `event_commands`, where you can add custom bash commands to be run on certain events. _pre_interaction_ runs before you talk to Alexa/Assistant, _post_interaction_ after it's finished. You could do your web calls here.
Add your own hooks where necessary in the [/src/main.py](https://github.com/xtools-at/AssistantPi/blob/master/src/main.py). 

Let me know how it goes :)

eouia, 2017-08-15:
Thanks for your reply. I'll close this.

closed 2017-08-15 by eouia