xtools-at/AssistantPi #19: Doesn't connect to google

kronflux, 2017-07-23:
Hey! I'm using a PineA64 board, which I've managed to get AlexaPi working on.
The Alexa side works perfectly. No issues, in fact better than I thought it would.
Google, however, locks the whole thing up.
Upon trying to run the assistant manually using:
`/opt/AlexaPi/env/bin/python` -m googlesamples.assistant --credentials /etc/opt/AlexaPi/assistant_credentials.json
I get:
`INFO:root:Connecting` to embeddedassistant.googleapis.com
which sits there indefinitely.
Any thoughts on diagnosing/debugging this?

(just as an FYI, I'm using https://github.com/swkim01/RPi.GPIO-PineA64 which replaces the standard RPi.GPIO, but I can't imagine that has anything to do with it. Especially where Alexa works flawlessly)

xtools-at, 2017-07-25:
Hi @kronflux ,

A little background first :) AssistantPi is basically AlexaPi, with [Google Assistant Python SDK 0.2.1](https://github.com/googlesamples/assistant-sdk-python/tree/0.2.1) hooked into it.
The installer runs the AlexaPi setup and then performs the necessary steps for installing Goggle Assistant seperately, and does the audio config for both.
When you fire it up, the AlexaPi script starts the Google Assistant process in the background too, and communicates with it via Pipes. So this is no seamless integration, but much more a proof of concept.

To finally answer your question :) I've mentioned the Google SDK version 0.2.1 specifically - I haven't updated to 0.3 yet as they introduced new libraries to communicate with Google Services. Those only work on certain Chip architectures (arm7 I think, you can find it in their repo's issues), which even breaks support for Raspberry Pi Zero (W), whereas not introducing too much other useful stuff...
Try grab the 0.2.1 Google SDK Release, install it standalone according to the README of _that_ version (!) and try to debug it that way. Tbh, I think this won't work as Google made this sample client we're using specifically for RPi :(

kronflux, 2017-07-25:
To be entirely clear, I have had the google assistant working previously using their tutorial on this device.
What I didn't have, unfortunately, was hotword detection, which is what drew me to this.

That being said, this clears things up a lot for me. Thanks!
I have almost no experience with modifying python scripts, so I was mostly looking for a quick and easy solution to getting voice control to google assistant, rather than requiring a button press :)

xtools-at, 2017-07-25:
You could try getting the latest standalone Google SDK, it supports voice control since 0.3 as well!

closed 2018-04-16 by xtools-at