xtools-at/AssistantPi #20: Custom actions possible?

mmgill, 2017-07-27:
I really liked what you guys have done here but had one issue that I was hoping to get more info on.

Using the standard google sdk I was able to create custom actions based on keywords (action.py) . (Ex. - "OK Google, turn on TV"; action.py would fire a subprcoess irsend command; tv would turn on)

Is it possible to implement custom actions/GPIO using Google/Alexa in this program? 

class IRRemote(object):
        def __init__(self, say, command):
        self.say = say
        self.command = command

    def run(self, voice_command):
        if self.command == "power":
            self.say("TV Power")
            subprocess.call(["irsend", "SEND_ONCE", "samsung", "KEY_POWER"])
        elif self.command == "mute":
            self.say("Muting TV")
            subprocess.call(["irsend", "SEND_ONCE", "samsung", "KEY_MUTE"])
        elif self.command == "switch to tv":
            self.say("Input 1")
            subprocess.call(["irsend", "SEND_ONCE", "hdmi", "KEY_PROG1"])
        elif self.command == "switch to input":
            self.say("Input2")
            subprocess.call(["irsend", "SEND_ONCE", "hdmi", "KEY_PROG2"])
        elif self.command == "switch to nexus":
            self.say("Input 3")
            subprocess.call(["irsend", "SEND_ONCE", "hdmi", "KEY_PROG3"])
        else:
            logging.error("Error identifying power command.")
            self.say("Sorry I didn't identify that command")
            

**Python 2 & 3 releases (`python2 --version` and `python3 --version`):**
```

```

**Description of problem:**
```

```

**Expected:**
```

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

xtools-at, 2017-07-28:
Hi @mmgill 
AssistantPi is using it's own instance of [Google Assistant Python SDK 0.2.1](https://github.com/googlesamples/assistant-sdk-python/tree/0.2.1) (find the fork that's installed [here](https://github.com/xtools-at/assistant-sdk-python)), which is downoaded to `/opt/AlexaPi/src/assistant-sdk-python`. If the custom actions don't require a newer version of the SDK, you can try doing your changes there.

I haven't updated to 0.3.x yet as they've 
* changed structure and functionality in a way that most of my integration would need to bee completely rewritten
* introduced new libraries to communicate with Google Services which only work on certain chip architectures and break support for Raspberry Pi Zero (W)
* not introduced too much other useful stuff (voice recognition and timers...).
When the SDK matures, I'll put together AssistantPi 2.0 ;)

mmgill, 2017-07-28:
Thanks for the info.  I did use the https://github.com/google/aiyprojects-raspbian repo to get GA running on the pi which has integrated custom voice commands using action.py.  Might have to look for other options to handle my situation, rasp pi3.  

Thanks!

closed 2017-08-20 by xtools-at