xtools-at/AssistantPi #23: Suggestion: Add mycroft AI to support home assistant devices / hue lights

5andr0, 2017-09-05:
Mycroft is the only pi sdk that supports home assistant lights.
To control lights with the alexa/gassist sdk you would have to own the genuine devices like alexa echo or google home to discover the lights.

IFTTT is too slow and you have to setup a new applet for every light.

[aiyprojects-raspbian](https://github.com/google/aiyprojects-raspbian/blob/master/src/action.py) has makers in [src/action.py](https://github.com/google/aiyprojects-raspbian/blob/master/src/action.py) to control hue lights, but that's not integrated in the basic assistant-sdk-python, which is used here.

xtools-at, 2017-09-07:
Hi,

I've already looked into this when I've started this project :D To run Cortana, you have to use MS IoT Core, which, according to my findings, can't be installed alongside Raspbian, but only as a replacement. If you've found a nicer way to get things going, please let me know!

I've red recently that Amazon and MS are jumping ship together and we'll be able to ask Cortana through Alexa and vice versa by the end of the year.

LegoChicken, 2017-09-07:
Wasn't the original issue about https://mycroft.ai/ not Microsoft?

xtools-at, 2017-09-07:
oops I wasn't aware of this, thought he'd be just kidding :D 

this should be doable, but I'm lacking the time. In particular, integrating Mycroft's setup and authentication into AssistantPi's setup is the more complicated thing, just hooking it up to the hotword recognition shouldn't be that much of an issue.
and then there's the hotword - I guess pocketsphinx doesn't come with "Mycroft" in their language model, I guess users will have to add that for themselves :)

xtools-at, 2017-09-07:
I've just have taken a closer look at mycroft core.
it comes with it's own hotword recognition, but there's a cli interface I could tap in instead.
authentication is done on the first call to the API via a digit code mycroft "tells" you and you enter it online - so nothing a setup script has to handle.

If you're comfortable with
- Installing and setting up Mycroft Core on your Pi
- Turning it on in AssistantPi config file
- Adding "Mycroft" Hotword to pocketsphinx language model
yourself, I could try to get to this when I've got some spare time.