xtools-at/AssistantPi #25: Setup.sh fails during python-vlc

shoemakn, 2017-10-23:
This is on latest Raspbian Stretch Lite (Jessie no longer available after WiFi exploit was patched), released 2017-09-07, kernel v4.9

**Python 2 & 3 releases (`python2 --version` and `python3 --version`):**
Python 2.7.13
Python 3.5.3

**Description of problem:**
Initial run of /opt/AlexaPi/src/scripts/setup.sh fails at the following:
```
Collecting python-vlc from git+https://github.com/oaubert/python-vlc.git#egg=python-vlc (from -r ./requirements.txt (line 3))
  Cloning https://github.com/oaubert/python-vlc.git to /tmp/pip-build-IFM6d6/python-vlc
    Complete output from command python setup.py egg_info:
    /usr/lib/python2.7/dist-packages/setuptools/dist.py:342: UserWarning: The version specified ('%s"') is an invalid version, this may not work as expected with newer versions of setuptools, pip, and PyPI. Please see PEP 440 for more details.
      "details." % self.metadata.version
    running egg_info
    creating pip-egg-info/python_vlc_generator.egg-info
    writing pip-egg-info/python_vlc_generator.egg-info/PKG-INFO
    writing top-level names to pip-egg-info/python_vlc_generator.egg-info/top_level.txt
    writing dependency_links to pip-egg-info/python_vlc_generator.egg-info/dependency_links.txt
    writing manifest file 'pip-egg-info/python_vlc_generator.egg-info/SOURCES.txt'
    error: package directory 'generator/generate' does not exist
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-IFM6d6/python-vlc/
```

**Steps to reproduce:**
1. Fresh Raspbian Stretch Lite install
2. Connect to network
3. Fresh git clone of AssistantPi to /opt/AlexaPi per installation intructions
4. Run sudo /opt/AlexaPi/src/scripts/setup.sh

xtools-at, 2017-10-23:
Thanks for reporting, I'm currently abroad and check on it when I'm back. You can try installing the missing dependency from the error message before running the setup.
Please report back if you run into any problems!

shoemakn, 2017-10-23:
Thanks!

I also tried installing on a fresh install of Jessie (I found the July release in my Downloads folder) with the same results.

Will try discerning the appropriate dependency and pre-installing it.

shoemakn, 2017-10-23:
It would appear this is a problem with the dependency itself: [python-vlc](https://github.com/oaubert/python-vlc/issues/34)

shoemakn, 2017-10-23:
Installed dependency: python-vlc via the latest commit _prior_ to the AssistantPi's commit date: [sha0b9da32bfb6ad6ed5296fa1cd967e0f5d101cbf1](https://github.com/oaubert/python-vlc/tree/0b9da32bfb6ad6ed5296fa1cd967e0f5d101cbf1)

Also had to manually install pip via [get-pip.py](https://bootstrap.pypa.io/get-pip.py) script to get the process going.

Initially had some issues with the Amazon/Alexa authorization

Encountered 3 errors during GoogleAssistant build, all pertaining to bdist_wheel:
[image]

Now the install says it was successful, but I apparently need to finish some audio configuration (at least)

shoemakn, 2017-10-24:
yeah.  nothing works as documented, I'm afraid.  I've tried both vlc and sox for audio.  I can successfully record and play back audio using arecord and aplay, but despite following all of the instructions I've found here and in the wiki, I've hit a brick wall.

cemtes, 2017-10-27:
Shoemakn, I'm encountering the same error with the python-vlc, but I'm totally new to linux. Could you go in to a bit of detail about how you solved this?  Thanks.

shoemakn, 2017-10-27:
**cemtes**, I wish I could help... but I haven't been able to resolve the issue myself.

It seems like several of the dependencies have some issues with their own setup scripts, so you can follow **xtools-at**'s advice above and try to find a way to get them installed before running the AssistantPi setup script, but your mileage may vary.  Best wishes & good luck!

cemtes, 2017-10-27:
Were you able to actually get it fully installed? That's the part I'm stuck on, and don't know how to install specific dependencies with the proper commit as you mentioned in your post.

shoemakn, 2017-10-27:
> **cemtes**, I wish I could help... but I haven't been able to resolve the issue myself.

nickexile, 2017-10-27:
Try running the following command to install python-vlc
`sudo pip2 install python_vlc `
That should install the dependency for you using python2

lucio97, 2017-10-28:
@nickexile solution work, thank you so much :D

cemtes, 2017-10-29:
@nickexile  thanks a ton. That worked.

cemtes, 2017-10-29:
@shoemakn  For what it's worth. I finally got it working. I had to do a fresh install of raspbian withOUT desktop. Then went through all the install. It kept pausing at various parts so I had to run the setup multiple times to get all the components. When I first ran it I was getting an error that it couldn't find "pulse" as an input so I had to go into the config.txt and change input from "pulse" to "mic" and it started accepting input. But then I wasn't getting any output. It turns out I didn't have pulseaudio installed. After a manual installation of pulse audio it seems to be working.

shoemakn, 2017-10-29:
@cemtes my install still fails during the bdist_wheel build process.  Did yours go without a hitch there, or did you have an additional dependency install?

Thanks @nickexile for the vlc install

PR #29 (2018-03-26): Use PyPI version for python-vlc
The github repo for python-vlc has a structure that will prevent its direct install from the requirements (the actual setup.py for the module is in generated/2.2). Use the PyPI version instead.

Closes #25
closed 2018-03-26 by xtools-at