xtools-at/AssistantPi #27: Google Wakeword starts and ends conversation immediately.

nickexile, 2017-10-27:
**Python 2 & 3 releases (`python2 --version` and `python3 --version`):**
```
Python 2.7.13
Python 3.5.3
```

**Description of problem:**
```
Google assistant starts with Wake word, then immediately finishes
```

**Expected:**
```
I expected google to let me speak before ending the conversation :)
```

**Steps to reproduce:**

1.  Run AssistantPi
2. Use Google Wakeword
3. Look at output

**Traceback (if applicable):**
```
2017-10-27 14:21:59 DEBUG: Setting up playback handler: VlcHandler
2017-10-27 14:21:59 INFO: Checking Internet Connection ...
2017-10-27 14:21:59 DEBUG: Starting new HTTPS connection (1): api.amazon.com
2017-10-27 14:22:04 DEBUG: https://api.amazon.com:443 "GET /auth/o2/token HTTP/1.1" 404 29
2017-10-27 14:22:04 INFO: Connection OK
2017-10-27 14:22:04 INFO: AVS token: Requesting a new one
2017-10-27 14:22:04 DEBUG: Starting new HTTPS connection (1): api.amazon.com
2017-10-27 14:22:05 DEBUG: https://api.amazon.com:443 "POST /auth/o2/token HTTP/1.1" 200 960
2017-10-27 14:22:05 INFO: AVS token: Obtained successfully
2017-10-27 14:22:05 DEBUG: Stopping audio play
2017-10-27 14:22:05 DEBUG: Playing audio: /opt/AlexaPi/src/resources/hello.mp3
2017-10-27 14:22:05 DEBUG: Player State: State.Opening
2017-10-27 14:22:05 DEBUG: Player State: State.Playing
2017-10-27 14:22:05 DEBUG: Started play. 
2017-10-27 14:22:08 DEBUG: Player State: State.Ended
2017-10-27 14:22:08 DEBUG: Finished play. 
INFO: cmn_live.c(120): Update from < 41.00 -5.29 -0.12  5.09  2.48 -4.07 -1.37 -1.78 -5.08 -2.05 -6.45 -1.42  1.17 >
INFO: cmn_live.c(138): Update to   < 42.70 -2.48 -5.29 -1.06 -4.24 -1.86  0.48 -8.38 -1.62  0.09 -3.25 -3.91 -1.07 >
INFO: kws_search.c(656): kws 1.51 CPU 0.702 xRT
INFO: kws_search.c(658): kws 4.83 wall 2.245 xRT
2017-10-27 14:22:10 DEBUG: Triggered with hotword: **google**
2017-10-27 14:22:10 DEBUG: Google Assistant Hotword(s): google
2017-10-27 14:22:10 INFO: Starting Assistant conversation
2017-10-27 14:22:10 INFO: Assistant conversation finished

```

**Additional info:**
```
You can see that the conversation starts and ends within the same second.
```

nickexile, 2017-10-27:
It looks like it is erroring on the line(358) of main.py:
`p.expect('Recording audio .*')`

nickexile, 2017-10-27:
Also the following worked for me:
`/opt/AlexaPi/env/bin/python -m googlesamples.assistant --credentials /etc/opt/AlexaPi/assistant_credentials.json`

wilhelmhell, 2017-12-17:
Got the same issue. Standalone assistant works fine but through AlexaPi it's not recording or playing okgoogle.mp3. Any ideas?

blackwiz4rd, 2018-03-13:
I am having the same issue, have you solved it?

xtools-at, 2018-04-16:
may be related to [this](https://github.com/googlesamples/assistant-sdk-python/issues/230) and #17