xtools-at/AssistantPi #30: [ACTION REQUIRED] Update Certificate authorities for your AVS client

futurepr0n, 2018-03-15:
Thank you for reporting an issue with **AssistantPi**.

Please make sure you've updated to the latest version using
```
sudo /opt/AlexaPi/src/scripts/update.sh
```
This updates both AssistantPi and the [tweaked Assistant SDK](https://github.com/xtools-at/assistant-sdk-python) without having you to go through the installation process again.

Please check out these resources before submitting an issue:
- [AlexaPi Wiki](https://github.com/alexa-pi/AlexaPi/wiki/)
- [AlexaPi Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/)
- [AlexaPi Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging)
- [Google Assistant Troubleshooting](https://developers.google.com/assistant/sdk/prototype/getting-started-pi-python/troubleshooting)
- [Google Assistant Audio Configuration](https://developers.google.com/assistant/sdk/prototype/getting-started-pi-python/configure-audio)

In order to help troubleshooting, be sure to include the following information:

**Python 2 & 3 releases (`python2 --version` and `python3 --version`):**
```

```

**Description of problem:**
```

```

**Expected:**
```

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```
Just wondering, I received an email from Amazon that requests I update certificate authorities for my AVS client by June 15 2018. Is this going to impact the AlexaPi part of AssistantPI - What should we do to ensure compatibility after that date - any advisement?
```

xtools-at, 2018-03-26:
this project is a mashup of AlexaPi and the Google Assistant SDK, can you check how the AlexaPi project is handling that?

futurepr0n, 2018-04-04:
Hmm thanks for the advisement! I will do that. By the way, I do not have an update.sh script at /opt/AlexaPi/src/scripts/ Is this maybe due to me having an older version? All I have is:

```

pi@raspberrypi:/opt/AlexaPi/src/scripts $ ls -al
total 44
drwxr-xr-x 4 root root 4096 Nov  1 13:49 .
drwxr-xr-x 5 root root 4096 Nov  1 13:49 ..
-rw-r--r-- 1 root root  753 Nov  1 13:49 AlexaPi.service
-rw-r--r-- 1 root root  260 Nov  1 13:49 asound.example.conf
drwxr-xr-x 4 root root 4096 Nov  1 13:49 inc
-rwxr-xr-x 1 root root 1925 Nov  1 13:49 initd_alexa.sh
-rwxr-xr-x 1 root root  464 Nov  1 13:49 monitorAlexa.sh
-rw-r--r-- 1 root root 1125 Nov  1 13:49 setup.bat
-rwxr-xr-x 1 root root 7001 Nov  1 13:49 setup.sh
drwxr-xr-x 2 root root 4096 Nov  1 13:49 unit-overrides
pi@raspberrypi:/opt/AlexaPi/src/scripts $

```

xtools-at, 2018-04-16:
yes, you should do a reinstall, you're on an ancient version :) but maybe not right away, since #31 isn't merged yet, and I'm planning on updating Assistant SDK as well

closed 2018-04-16 by xtools-at