xtools-at/AssistantPi #32: Error when running sudo ./AlexaPi/src/scripts/setup.sh

u1608, 2019-07-21:
Kindly help,
Setup stops here.Thanks in advance.cheers

Creating a user to run AlexaPi under ... user already exists. That's cool - using that.
gpio:x:997:pi,alexapi
audio:x:29:pi,alexapi
Synchronizing state of AlexaPi.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable AlexaPi
Would you like to also install Airplay support (Y/n)? y
Hit:1 http://archive.raspberrypi.org/debian buster InRelease                                           
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease                                        
Hit:3 https://deb.nodesource.com/node_10.x buster InRelease                      
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package vlc-nox is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  vlc-bin

E: Package 'vlc-nox' has no installation candidate

rjgladish, 2020-01-18:
Look here for clues to resolve vlc dependency issues.
https://www.raspberrypi.org/forums/viewtopic.php?t=195221

kahupihu, 2020-12-19:
It took me a lot of time but I fixed it finally...
1. Open terminal.
2. Paste  "  sudo nano /opt/AlexaPi/src/scripts/inc/os/debian.sh  ".
3. Replace vlc-nox with vlc-bin.
4. Ctrl + X
5. Y
6. Enter
Done!

EmadMokhtar, 2021-01-26:
> It took me a lot of time but I fixed it finally...
> 
> 1. Open terminal.
> 2. Paste  "  sudo nano /opt/AlexaPi/src/scripts/inc/os/debian.sh  ".
> 3. Replace vlc-nox with vlc-bin.
> 4. Ctrl + X
> 5. Y
> 6. Enter
>    Done!

What about creating a PR for this fix?